dorssel/usbipd-win #800: usbipd: error: Unable to run 'usbip' client tool.

TK1987, 2023-12-14:
Hi there.

While trying to attach an usb device, i got the following message:
```
PS C:\Windows\system32> usbipd attach -b 3-18 -w
usbipd: info: Using WSL distribution 'Debian' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.
```

Version: `usbipd-win 4.0.0`
OS: `Debian 12 Bookworm`
Kernel: `5.15.133.1-microsoft-standard-WSL2+`

I then uninstalled version 4.0.0 and reinstalled version 3.2.0, which works fine.

Regards, Thomas

dorssel, 2023-12-14:
Could you run the diagnostics as given in https://github.com/dorssel/usbipd-win/issues/793?

TK1987, 2023-12-14:
```
PS C:\Windows\System32> wsl --user root '/media/c/Program Files/usbipd-win/wsl/usbip' version
usbip (usbip-utils 2.0)

PS C:\Windows\System32> wsl --list --all --verbose
  NAME      STATE           VERSION
* Debian    Running         2
```

I have changed the root directory of the [automount settings](https://learn.microsoft.com/en-us/windows/wsl/wsl-config#automount-settings) to `/media`. This seems to be the problem.
```
thomas@PC01:~$ cat /etc/wsl.conf
[boot]
systemd=true

[automount]
root=/media/

[user]
default=thomas
```

After I removed the change, it also works in version 4.0.0. Is there a way to make this work with the change?

dorssel, 2023-12-14:
Ah! Thanks for the report.

I was unaware of the `[automount]` options. Obviously, when `enabled=false` it is not going to work. But `root=` should be supported. I will add support for that. For now, either run the `usbip` client tool yourself, or remove the `root=` option as a workaround.

dorssel, 2023-12-15:
I've implemented this in #802, you can find the (working version) installer here: https://github.com/dorssel/usbipd-win/actions/runs/7216638955?pr=802

Could you test it?

TK1987, 2023-12-15:
Wow, that was fast.

> Could you test it?

Looks good.

> Obviously, when `enabled=false` it is not going to work. 

Even that works as long as the C drive is mounted somewhere.

Great Job!

closed 2023-12-15 by TK1987

biggestsonicfan, 2024-01-14:
I installed OpenSUSE Tumbleweed as my distro and changed my `[automount]` options as the "SUSE way" doesn't use `/mnt/` but rather `/run/media/`. In combination with `wsl-usb-gui` I can see my device via `lsusb`, bravo!

dorssel, 2024-01-14:
@biggestsonicfan 
The current master build has a fix that now honors the automount options, you can find it here: https://github.com/dorssel/usbipd-win/actions/runs/7472527220. Can you confirm that it works with the default `/run/media` for your distro?

biggestsonicfan, 2024-01-15:
@dorssel 
Can confirm it works, though somewhere along the line `wsl-usb-gui` got broken with a `usbipd: error: The 'wsl' subcommand has been removed. Learn about the new syntax at https://learn.microsoft.com/windows/wsl/connect-usb#attach-a-usb-device` error. So I am just binding via powershell and it's there.

Can't quite get my phone mounted via `simple-mtpfs` without `root` yet though. The udev rules `wsl-usb-gui` seemed to create apply to ubuntu udev and not suse udev 😖

dorssel, 2024-01-15:
@biggestsonicfan 
The latest release of `wsl-usb-gui` supports the new syntax, maybe you want to upgrade that too.

PR #802 (2023-12-16): Add support for WSL [automount] configuration
Fixes #800