dorssel/usbipd-win #805: Installation Unsuccessful - Windows Server 2022 - 21H2

zachgoold, 2023-12-19:
When trying to install the MSI I am getting a "The operation ended prematurely because of an error. Your computer was not modified.". I am not sure how to get this resolved. Any help is appreciated!

Logs:
[usbipd-win.log](https://github.com/dorssel/usbipd-win/files/13718246/usbipd-win.log)

dorssel, 2023-12-19:
The log states:

> DIFXAPP: INFO:   Could not open file C:\Windows\System32\DriverStore\FileRepository\vboxusb.inf_amd64_5de009937a475578\vboxusb.inf. (Error code 0x3: The system cannot find the path specified.)
> ...
> DIFXAPP: ERROR:  PnP Install failed. (Error code 0x3EE: The volume for a file has been externally altered so that the opened file is no longer valid.)
> ...
> DIFXAPP: INFO:   Attempting to rollback ...

This means the (required) driver failed to install. This is due to a corrupt driver database. See https://github.com/dorssel/usbipd-win/issues/643 on how it was solved for another user.

dorssel, 2024-02-04:
4.1.0 improved driver installation

closed 2024-02-04 by dorssel