dorssel/usbipd-win #806: WSL can't list

stoffej, 2023-12-20:
Running on Win11
[image]

Cant list the devices in windows.(tried both ip:s)
[image]
The ip is pingable from wsl, and port is open.
Works with open a server on 3240 and connect to it using telnet on the same port.

Usbipd in debug mode.
[image]

dorssel, 2023-12-21:
What distro are you using and what's the output of `usbip version`.

stoffej, 2023-12-21:
[image]
[image]
[image]
[image]

It works on my colleagues computer. But not mine. And no traffic is generated from the "usbip list ..."
Can something with the network settings block this?
Firewall in windows is disabled

dorssel, 2023-12-21:
What `usbip` client package is installed? What is the distribution? Since 4.0.0, `usbipd-win` ships its own `usbip` client tools. What happens if you simply do `usbipd attach --wsl` on the Windows side?

stoffej, 2023-12-21:
Thanks for the quick reply.
How do find info about this? "What usbip client package is installed? "
[image]
The distrubution is
[image]

dorssel, 2023-12-21:
OK, so it is not `usbip`, that is OK. And it is not the `usbipd` service (as that would have given a warning too). The connection between WSL and the service is also working properly, as that is checked before the attach itself is even attempted.

All that remains is that Windows is not able to change the driver "on-the-fly". Try binding the device with `usbipd bind --force`.

stoffej, 2023-12-21:
Yes , but ping works and I can manually create a server on port 3240 on windows and connect to it from wsl.
Leaning to reinstall windows due to some hickup with network, but thats the last resort. Wanted to check with you if it could be something with uspipd <-> usbip first.

dorssel, 2023-12-22:
In the past I've seen issues with the version of `usbip` client tooling. Until 4.0.0 it was difficult to get the right client tools installed. However, your distro is modern (22.04), so that isn't it. I suggest you try to do a repair install or reinstall of `usbipd-win` first.

stoffej, 2023-12-22:
I have tried that and, is there a special usbip that is installed ?

dorssel, 2023-12-22:
It is installed on the Windows side, and it is used within WSL as `/mnt/c/Program Files/usbipd-win/wsl/usbip`.

dorssel, 2024-02-04:
4.1.0 has improved WSL support, could you try that?

stoffej, 2024-02-05:
I reinstalled my pc and now usbipd works again.
So something else was preventing it to work.
Suggest that this Issue shall be closed

closed 2024-02-05 by dorssel