dorssel/usbipd-win #807: The Rockchip RV1126 cannot connect to WSL2 as an ADB device.

johnzhy, 2023-12-21:
usbipd-win version: 4.0.0+182.Branch.master.Sha.2ffe37ec799b9e73eb9d23d051d980fefb616ce1
wsl2: ubuntu20.04  5.15.133.1-microsoft-standard-WSL2

rv1126 as adb device:
When I execute this command: usbipd attach --wsl -b 8-4

log:
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 105 ms
info: Usbipd.ConnectedClient[1]
      Client 172.27.199.146 claimed device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Trapped SET_CONFIGURATION: 1
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
info: Usbipd.ConnectedClient[2]
      Client 172.27.199.146 released device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: Usbipd.Server[1000]
      connection close: The operation was canceled.
dbug: Usbipd.Server[1000]
      connection closed

The RV1126 is still connected to the host machine.
wsl2 execute: lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

rv1126 as ntb device:
execute this command: usbipd attach --wsl -b 8-4

log:
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 1274 ms
info: Usbipd.ConnectedClient[1]
      Client 172.27.199.146 claimed device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Trapped SET_CONFIGURATION: 1

wsl2 execute: lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 020: ID 2207:0006 Fuzhou Rockchip Electronics Company rk3xxx
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

dorssel, 2023-12-22:
> USBSUP_XFER_DNR -> ETIME -> -62

Can you get a pcap? See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting#usb-capture.

johnzhy, 2023-12-22:
[FILE.zip](https://github.com/dorssel/usbipd-win/files/13751385/FILE.zip)

dorssel, 2023-12-22:
That log file only contains attaching the device. No data at all (and therefore also not the USBSUP_XFER_DNR failure).

johnzhy, 2023-12-23:
Sorry, I don't know how to obtain the correct logs. 
I followed the tutorial and executed the following command:
Open PowerShell as an administrator
usbipd server Logging:LogLevel:Default=Trace "usbipd:PcapNg:Path=D:\FILE.pcapng"
usbipd attach --wsl -b 8-4

dorssel, 2023-12-23:
That is the correct way. And the logs you returned are correct. However, when you captured the logs, the error did not occur. Try to repeat the situation where you got `USBSUP_XFER_DNR -> ETIME -> -62` while capturing logs.

johnzhy, 2023-12-23:
**this is log:**
`usbipd server Logging:LogLevel:Default=Trace "usbipd:PcapNg:Path=D:\rv1126.pcapng"`

dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = 'D:\rv1126.pcapng'
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:SnapLength = unlimited
dbug: Usbipd.Server[1000]
      usbipd:Port = 3240
dbug: Usbipd.Server[1000]
      4.0.0+182.Branch.master.Sha.2ffe37ec799b9e73eb9d23d051d980fefb616ce1
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Program Files\usbipd-win\
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 107 ms
info: Usbipd.ConnectedClient[1]
      Client 172.27.199.146 claimed device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Trapped SET_CONFIGURATION: 1
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
info: Usbipd.ConnectedClient[2]
      Client 172.27.199.146 released device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: Usbipd.Server[1000]
      connection close: The operation was canceled.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 104 ms
info: Usbipd.ConnectedClient[1]
      Client 172.27.199.146 claimed device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Trapped SET_CONFIGURATION: 1
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> ETIME -> -62
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
info: Usbipd.ConnectedClient[2]
      Client 172.27.199.146 released device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: Usbipd.Server[1000]
      connection close: The operation was canceled.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.PcapNg[1000]
      Flushing
info: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
dbug: Microsoft.Extensions.Hosting.Internal.Host[3]
      Hosting stopping
dbug: Microsoft.Extensions.Hosting.Internal.Host[4]
      Hosting stopped

**This is pcapng file:**
[rv1126.zip](https://github.com/dorssel/usbipd-win/files/13757794/rv1126.zip)

I executed the command `usbipd attach --wsl -b 8-4` twice

dorssel, 2023-12-23:
The interesting part:

```
17	0.1853787	host	8.4.0	USB	64	SET CONFIGURATION Request	Operation now in progress (-EINPROGRESS)
18	0.1863332	8.4.0	host	USB	64	SET CONFIGURATION Response	Success
19	0.1865966	host	8.4.0	USB	64	GET DESCRIPTOR Request STRING	Operation now in progress (-EINPROGRESS)
20	0.1868967	8.4.0	host	USB	64	GET DESCRIPTOR Response	Timer expired (-ETIME)
```

The device seems non-responsive right after setting the configuration. This is not strange, some devices just require a bit of "setup time". What *is* weird is that this leads to ETIME. As you can see from the time-stamps, there is just microseconds between the request and ETIME, so this probably is *not* a timer.

ETIME comes from the VirtualBox USBSUP_XFER_DNR (DNR == Did Not Respond). The translation to ETIME is mine. Maybe I should translate it to STALL, or so. I'll make a special build with that change, so you can see if that makes a difference.

dorssel, 2023-12-23:
Can you try the installer from https://github.com/dorssel/usbipd-win/actions/runs/7308106135?pr=812 and run another log session?

johnzhy, 2023-12-25:
logfile：
[rv11262.zip](https://github.com/dorssel/usbipd-win/files/13763186/rv11262.zip)

log：
usbipd server Logging:LogLevel:Default=Trace "usbipd:PcapNg:Path=D:\rv11262.pcapng"
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = 'D:\rv11262.pcapng'
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:SnapLength = unlimited
dbug: Usbipd.Server[1000]
      usbipd:Port = 3240
dbug: Usbipd.Server[1000]
      4.0.1-PullRequest0812.9+Branch.pull-812-merge.Sha.a009d3a5016f311bc68862e29b193aca07b7f687
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\KF2
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 105 ms
info: Usbipd.ConnectedClient[1]
      Client 172.27.199.146 claimed device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Trapped SET_CONFIGURATION: 1
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
info: Usbipd.ConnectedClient[2]
      Client 172.27.199.146 released device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: Usbipd.Server[1000]
      connection close: A task was canceled.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.27.199.146
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 106 ms
info: Usbipd.ConnectedClient[1]
      Client 172.27.199.146 claimed device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Endpoint created
dbug: 8-4.0[1000]
      Trapped SET_CONFIGURATION: 1
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: 8-4.0[1000]
      USBSUP_XFER_DNR -> EPIPE -> -32
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
info: Usbipd.ConnectedClient[2]
      Client 172.27.199.146 released device at 8-4 (USB\VID_2207&PID_0006\7&1B1B1ACF&1&4).
dbug: Usbipd.Server[1000]
      connection close: The operation was canceled.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.PcapNg[1000]
      Flushing
info: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
dbug: Microsoft.Extensions.Hosting.Internal.Host[3]
      Hosting stopping
dbug: Microsoft.Extensions.Hosting.Internal.Host[4]
      Hosting stopped

dorssel, 2023-12-26:
It gets (only marginally) further. But this time (right after the EPIPE errors) the device responds with an invalid reply!

```
Frame 32: 319 bytes on wire (2552 bits), 319 bytes captured (2552 bits) on interface USB, id 0
USB URB
    [Source: 8.4.0]
    [Destination: host]
    URB id: 0x0000000000004643
    URB type: URB_COMPLETE ('C')
    URB transfer type: URB_CONTROL (0x02)
    Endpoint: 0x80, Direction: IN
    Device: 4
    URB bus id: 8
    Device setup request: not relevant ('-')
    Data: present ('\0')
    URB sec: 1703465032
    URB usec: 484430
    URB status: Success (0)
    URB length [bytes]: 255
    Data length [bytes]: 255
    [Request in: 31]
    [Time from request: 0.001130100 seconds]
    Unused Setup Header
    Interval: 0
    Start frame: 0
    Copy of Transfer Flags: 0x00000200, Dir IN
    Number of ISO descriptors: 0
STRING DESCRIPTOR
    bLength: 0
        [Expert Info (Error/Malformed): Invalid STRING DESCRIPTOR Length (must be 2 or larger)]
    bDescriptorType: 0x00 (unknown)
Leftover Capture Data [truncated]: 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
```

I think we can safely conclude this device (or at least in this mode) is not going to function. It seems to violate the USB standard. Please uninstall the test version and revert to the released 4.0.0. I will list this device as incompatible.

johnzhy, 2023-12-28:
I found that by modifying the usb configuration file of the rv1126, the device can be recognized by WSL.

`vi /etc/init.d/.usb_config`
add line, the file like this
```
usb_adb_en
usb_ntb_en
```
then, reboot device
using the command `usbipd attach --wsl -b 8-4` to attach usb device to wsl

Execute the command `sudo adb devices` on rv1126 will list ADB devices.

Hope it is helpful for others using rv1126.

dorssel, 2023-12-28:
@johnzhy 
Thanks for the report. I've added the device to https://github.com/dorssel/usbipd-win/wiki/Tested-Devices with a reference to your findings. Is your issue resolved now?

johnzhy, 2023-12-29:
yes, It's working properly

closed 2023-12-29 by dorssel