dorssel/usbipd-win #808: Device disconnects after attach (Updated to USBIPD V4.0.0)

jesgarmar, 2023-12-22:
Hello,

I've updated to the newest version of usbipd. I'm binding a device (RealSense camera) and when i try to attach to my WSL, the device disconnects and there is an error msg:
"usbipd: error: Failed to attach device). The "usb unplug sound" occurs and then the device is no longer detected in windows.

I was working with the older version but its not longer working. I'm attaching a picture and i can provide more information if needed. 

PD: The camera was working till last update of usbipd and i've updated the wsl with "wsl --update".

Thanks
[image]

dorssel, 2023-12-22:
The version of WSL is not important here. This problem arises from the driver switch (Windows-side) that needs to happen at the time of attach. It depends on whether the device is (or was) in use by Windows. This functionality hasn't changed between 3.2.0 and 4.0.0, but the included VirtualBox drivers were updated (which may be the cause). If it doesn't work after reboot (the usual Windows solution for almost everything), you may have to bind the device using `--force`.

jesgarmar, 2023-12-22:
> The version of WSL is not important here. This problem arises from the driver switch (Windows-side) that needs to happen at the time of attach. It depends on whether the device is (or was) in use by [...]

It worked after binding with '--force' param. Thanks!

closed 2023-12-22 by jesgarmar