dorssel/usbipd-win #809: Unhandled exeption in WSL list

lefebvresam, 2023-12-23:
Using Windows 10 Powershell in admin mode:

```
PS C:\Users\Sam Lefebvre> usbipd list
Unhandled Exception: System.AggregateException: One or more errors occurred. (DEVPKEY_Device_LocationInfo returned 'Port_#0001.MAC_Addr_62006067', expected form 'Port_#0123.Hub_#4567')
 ---> System.NotSupportedException: DEVPKEY_Device_LocationInfo returned 'Port_#0001.MAC_Addr_62006067', expected form 'Port_#0123.Hub_#4567'
   at Usbipd.ConfigurationManager.GetBusId(UInt32) + 0x207
   at Usbipd.UsbDevice.GetAll() + 0xff
   at Usbipd.CommandHandlers.Usbipd.ICommandHandlers.List(Boolean, IConsole, CancellationToken) + 0x36
   at Usbipd.Program.<>c__DisplayClass11_4.<<Run>b__39>d.MoveNext() + 0xa7
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Invocation.AnonymousCommandHandler.<InvokeAsync>d__6.MoveNext() + 0x1ca
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext() + 0xe8
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass17_0.<<UseParseErrorReporting>b__0>d.MoveNext() + 0xed
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass12_0.<<UseHelp>b__0>d.MoveNext() + 0xbf
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseVersionOption>b__0>d.MoveNext() + 0x1da
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass19_0.<<UseTypoCorrections>b__0>d.MoveNext() + 0x13b
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__18_0>d.MoveNext() + 0x193
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass16_0.<<UseParseDirective>b__0>d.MoveNext() + 0x101
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__5_0>d.MoveNext() + 0x194
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<CancelOnProcessTermination>b__1_0>d.MoveNext() + 0x14d
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Invocation.InvocationPipeline.<<InvokeAsync>g__FullInvocationChainAsync|2_0>d.MoveNext() + 0xf7
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Parsing.ParseResultExtensions.<InvokeAsync>d__0.MoveNext() + 0xc8
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at System.CommandLine.Parsing.ParserExtensions.<InvokeAsync>d__3.MoveNext() + 0xb5
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean) + 0x71
   at Usbipd.Program.Run(IConsole, ICommandHandlers, String[]) + 0x1eb1
   at usbipd!<BaseAddress>+0x3892f0
```

dorssel, 2023-12-23:
Thanks for the report. This is quite a unique find!

Can you open Windows Device Manager, and in "View" select  devices by connection?
Then try to locate this device, it must be somewhere under a USB root. I am interested in:
- what USB hub is this device connected to
- what driver is installed *for that hub*

For example:

[image]

Here I found the device with Port_#0004.Hub_#0005. In your case you're looking for a device that reports Port_#0001.MAC_Addr_62006067.

Then I want to know the drivers of the Hub the device is connected to (in my case, the "Generic USB Hub" right above it):

[image]

lefebvresam, 2023-12-23:
After one night, the problem seems to be solved (it's like the wine :-)). However I use an i-tec docking station with Windows on a MAC with bootcamp. I already had other stability issues like a screen that is turning off after each sleep session, a SD that dissapears surprisingly, or another screen that sometimes begin to flicker. My programmer looks like:

[image]

[image]

[image]

dorssel, 2023-12-23:
The "location" (which we use for BUSID mapping) is determined by the hub. What are the drivers of the *hub* (in your case the "USB-hoofdhub (USB3.0)"? What went wrong (initially) is that the Location is supposed to be of the format "Port_#xxxx.Hub_#yyyy", but for some reason it had a totally different format for the last (hub-) part.

In any case, I'll make a fix so at least it displays a better warning instead of just crash. But I would still like to know the root cause of the weird numbering. In the past I've seen similar issues with non-Microsoft hub drivers. Hence the interest in the driver of the hub.

lefebvresam, 2023-12-23:
[image]

dorssel, 2023-12-23:
Yes, now it is normal. If you ever see the error again, please report the driver at that time. I guess it is resolved for now.

lefebvresam, 2023-12-24:
I found the root cause. I have an audio USB device connected over Ethernet with USB server. When the device is on the network it will be connected automatically to my PC and when you do a listing then it invokes such message. By turning off the device at the TV, waiting for the disconnecting sound, doing a list, the error is gone.

lefebvresam, 2023-12-24:
[image]

dorssel, 2023-12-27:
That's very interesting!

You could, in principle, re-forward such a device. But this is probably going to be a) senseless (it already is a networked USB device, so you better connect it directly to the destination instead of doing multiple network hops), b) asking for even more timing problems (each network hop already introduces latencies beyond the USB spec), c) asking for more driver problems (USBIP relies on a properly functioning hub; I've seen many non-Microsoft hub drivers messing up).

All-in-all I don't think it is worth the trouble of trying to support this (networked) hub. However, it would be nice if `usbipd-win` wouldn't crash, of course. And even better if it would give a warning that such incompatible devices were found. I'll make a change for that.

lefebvresam, 2023-12-27:
Note that the network connection you mention is between a Windows pc and a WSL session running on the same PC. I was thinking about a scenario where you have a setup at a remote location in the building and you want to connect a programmer (STLinkV2 clone) via an USB Ethernet convertor (USB1000IP Startech.com) and do the programming from your PC. This could be a useful setup if your board must be connected to peripherals that are not nearby.

dorssel, 2023-12-27:
Doesn't the USB Ethernet Converter come with direct tooling for Linux? Why would you need to attach it to the host first, only to forward it again? I assume you could attach directly from WSL to the USB Ethernet Converter?

lefebvresam, 2023-12-27:
I have to investigate if you can run the same service in Linux WSL as you can run in Windows. I'm currently connecting the convertor to a sound mixing table (Behringer) and use the device over USB and Ethernet as a recording device in my PC to record multi track sound. Because the TV sound is also coming in trough a channel I can record TV audio in high quality. If the setup is so flexible that the service connects automatically to a programmer after starting WSL and the listing is in the same format I agree with your vision.

dorssel, 2024-01-01:
Could you try the installer from https://github.com/dorssel/usbipd-win/actions/runs/7374513166?pr=819 to confirm this is fixed?

lefebvresam, 2024-01-01:
New output:

```
C:\Users\Sam Lefebvre>usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
       08bb:2902  USB Audio CODEC, USB-invoerapparaat                           Incompatible hub
1-7    0000:0002  Onbekend USB-apparaat (verzoek voor apparaatdescriptor is...  Not shared
2-1    05ac:8233  NCM Composite Device                                          Not shared
2-2    05ac:8514  FaceTime HD Camera (Built-in)                                 Not shared
2-3    05ac:8262  USB-invoerapparaat                                            Not shared
2-4    05ac:8103  USB-invoerapparaat                                            Not shared
2-5    05ac:027b  USB-invoerapparaat, Apple Keyboard Internal USB, Apple US...  Not shared
2-6    05ac:8302  Apple Touch Bar                                               Not shared
2-7    05ac:8102  USB-invoerapparaat                                            Not shared
10-4   0d8c:0022  USB Advanced Audio Device, USB-invoerapparaat                 Not shared
10-7   05e3:0749  USB-apparaat voor massaopslag                                 Not shared
11-1   0c45:6366  8M Web Camera                                                 Not shared
12-2   046d:c52b  Logitech USB Input Device, USB-invoerapparaat                 Not shared
13-2   0403:6001  USB Serial Converter                                          Not shared
13-3   0922:1002  Ondersteuning voor USB-afdrukken, USB-apparaat voor massa...  Not shared
14-1   0403:6001  USB Serial Converter                                          Not shared
14-2   046d:c408  USB-invoerapparaat                                            Not shared
14-3   1a44:0870  Microsoft Usbccid-smartcardlezer (WUDF)                       Not shared

Persisted:
GUID                                  DEVICE
d6d2745b-2396-487a-ab79-79ad18c4a88a  STM32 STLink
```

dorssel, 2024-01-01:
Thanks for testing.

closed 2024-01-01 by dorssel

cryptonaut01, 2024-01-04:
[image]
Run into same issue with a WIFI device connected to an Etron USB Hub.
With version 4.0.1 the usbipd list output works again.

dorssel, 2024-01-04:
@cryptonaut01 
In your case, it seems like the drivers from Etron are incompatible, but the default Microsoft drivers would probably work. In device manager, you could try:

- right-click "Etron USB 3.0 Extensible Root Hub"
- select "Update Driver"
- select "Browse my computer for drivers"
- select "Let me pick from a list of available drivers on my computer"
- select Manufacturer: "(Standard USB HUBs)", Model: "USB Root Hub (USB3.0)"

In my experience, since 2015 it has not been necessary anymore to have specific manufacturer drivers. Instead, the default Microsoft drivers work fine, and in many cases even better.

PR #819 (2024-01-01): Fix crash for incompatible hubs
Fixes #809