dorssel/usbipd-win #810: how to hook up usbipd with wsl2 without hyper-v, need help

powerpun, 2023-12-23:
with hyper-v bridge mode, it works but how I could hook up usbipd and wsl2 without hyper-v?
like 
[experimental]
autoMemoryReclaim=gradual
networkingMode=mirrored
dnsTunneling=true
firewall=true
autoProxy=true

because hyper-v uses bridge mode and slow down the whole network speed.

dorssel, 2023-12-23:
Are you using `usbipd-win` version 4.0.0? (if not, you should...)

powerpun, 2023-12-23:
thanks for your kind and prompt reply. after the upgrade, still could not work, so, what should I do? add port forwarding?

usbipd: info: Using WSL distribution 'Debian.11' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 192.168.0.90 to reach the host.
WSL usbip: error: tcp connect
usbipd: error: Failed to attach device with busid '3-2'.

dorssel, 2023-12-23:
What is the output of
```
wsl --distribution Debian.11 --user root -- bash -c 'echo < /dev/tcp/192.168.0.90/3240'
``

powerpun, 2023-12-23:
PS C:\Windows\system32> wsl --distribution Debian.11 --user root -- bash -c 'echo < /dev/tcp/192.168.0.90/3240'
bash: connect: Connection refused
bash: line 1: /dev/tcp/192.168.0.90/3240: Connection refused

firewall problem?

dorssel, 2023-12-23:
Yes, or 192.168.0.90 is *not* the right address, or the `usbipd-win` service is not actually running.
But for the last one, you should have had a warning during `attach`, which you didn't.

Most likely firewall. `usbipd-win` does install some firewall rules. Try a repair install of `usbipd-win` first. Are you running any 3rd party security products?

powerpun, 2023-12-23:
I don't use 3rd party security products at all. 
with netstat -a -n result, found 
 TCP    [::]:3240              [::]:0                 LISTENING

so I wonder if we could use another port instead? let me reboot the computer and try again.
wsl version:

WSL version： 2.0.14.0
kernel version： 5.15.133.1-1
WSLg version： 1.0.59
MSRDC version： 1.2.4677
Direct3D version： 1.611.1-81528511
DXCore version： 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows version： 10.0.22621.2861

dorssel, 2023-12-26:
Reopening.

I now realize that originally you were using mirrored networking mode, yet `usbipd-win` tried to use a normal IP. It is correct that for mirrored networking mode the IP address of the host and WSL are the same. However, `usbipd-win` should have detected the mirrored networking mode and should have used 127.0.0.1 instead.

Now I wonder why the detection was wrong...

Could you please post the output of:
```
wsl --distribution Debian.11 --user root -- /usr/bin/wslinfo --networking-mode
```
for both networking modes (mirrored and not-mirrored)?

powerpun, 2023-12-30:
sorry for the delay cause I did not open github for a while.
/bin/bash: line 1: /usr/bin/wslinfo: No such file or directory
I think this caused the problem. but for the latest version of usbip we don't need to install anything in linux side?

dorssel, 2023-12-31:
Yes, I expected that much. It's been fixed in https://github.com/dorssel/usbipd-win/pull/817. Docker-desktop had the same problem. For some reason my own Ubuntu has it in both /usr/bin as well as /bin. But /bin is always present. Thanks for checking.

If you want, you can use the latest master build from https://github.com/dorssel/usbipd-win/actions/runs/7343009478 and switch back to mirrored mode. Or wait for the next release.

closed 2023-12-31 by dorssel