dorssel/usbipd-win #811: usbipd error: WSL kernel not USBIP capable

Chris2640BE, 2023-12-23:
Hi there,

A few weeks ago I reinstalled from scratch a Windows host (Medion Akoya E8715 MD60123 x64-based portable, Intel Core i3-5005U CPU @ 2.00GHz with 2 core’s, Windows 10 Home version 10.0.19045, 6GB memory, …) and installed Docker Desktop on it in order to run a Home Assistant in e Docker container.

In order to be able to connect Zigbee devices to HA I bought a few days ago a Zigbee slae.sh cc2652rb controller USB-stick. Because the stick was not recognised by the Windows-host I also installed the CP210x Universal Windows Driver (https://www.silabs.com/documents/public/software/CP210x_Universal_Windows_Driver.zip) as recommended by the HA-community (https://community.home-assistant.io/t/tutorial-to-upgrade-slaeshs-cc2652rb-zigbee-usb-stick/374306).

According to the General FAQ’s for Docker Desktop, it is not possible to pass through the USB stick to the HA-container because it requires support at the hypervisor level.

In order to solve this i installed uspipd-win on the Windows host, ran usbipd list in order to identify the device, shared it with the uspipd bind command using the BUSID I found using usbipd.

After receiving the message “usbipd: error: WSL kernel is not USBIP capable; update with ‘wsl --update’” I updated the Windows Subsystem fot Linux to version 2.0.14.

In order to be sure that the update was complete I repeated the update and got the message “The most recent version of Windows Subsystem for Linux is already installed”.

When I finally try to attach the device to the running HA-container with the command “isbipd attach --wsl --busid=<mybusid>” I get the error message “usbipd: error: WSL kernel is not USBIP capable; update with ‘wsl --update’”.

When I retry to update I once more get the message that the most recent version of Windows Subsystem for Linux already is installed, so it seams that I’m stuck.

In an effort to solve the problem I checked for updates and updated Docker Desktop to version 4.26.1 (131620) (Docker engine v24.0.7), but that didn’t solve the problem either.

Any idea’s about the reasons of this problem and, even better, how I can resolve it?

Chris

dorssel, 2023-12-23:
What is the output of `wsl --list --all --verbose`?

Chris2640BE, 2023-12-23:
The wsl --list  --all verbose command returns following answer:

   NAME                            STATE         VERSION
* docker-desktop-data     Running     2
   docker-desktop             Running     2

dorssel, 2023-12-23:
Can you give the output of
```
wsl --distribution docker-desktop-data --user root -- cat /sys/devices/platform/vhci_hcd.0/status
wsl --distribution docker-desktop --user root -- cat /sys/devices/platform/vhci_hcd.0/status
```

Chris2640BE, 2023-12-23:
Hi Frans,

 

The first command results in following output:

 

<3>WSL (27) ERROR: CreateProcessParseCommon:700: Failed to translate
C:\Windows\system32

<3>WSL (27) ERROR: CreateProcessParseCommon:737: getpwnam(root) failed 2

 

And the second gives:

 

hub port sta   spd  dev             sockfd   local_busid

hs   0000 004 000  00000000 000000  0-0

hs   0001 004 000  00000000 000000  0-0

hs   0002 004 000  00000000 000000  0-0

hs   0003 004 000  00000000 000000  0-0

hs   0004 004 000  00000000 000000  0-0

hs   0005 004 000  00000000 000000  0-0

hs   0006 004 000  00000000 000000  0-0

hs   0007 004 000  00000000 000000  0-0

hs   0008 004 000  00000000 000000  0-0

hs   0009 004 000  00000000 000000  0-0

hs   0010 004 000  00000000 000000  0-0

hs   0011 004 000  00000000 000000  0-0

hs   0012 004 000  00000000 000000  0-0

hs   0013 004 000  00000000 000000  0-0

hs   0014 004 000  00000000 000000  0-0

hs   0015 004 000  00000000 000000  0-0

 

Chris

capable (Issue #811)

 

<https://github.com/dorssel/usbipd-win/issues/811#issuecomment-1868299409> ,
3PD5AVCNFSM6AAAAABBAUNW3WVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTQNRYGI4TS
NBQHE> .
<https://github.com/notifications/beacon/A5NKN2WH4U7MPY6WQLUQMWLYK3PD5A5CNFS
M6AAAAABBAUNW3WWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT

dorssel, 2023-12-23:
`usbipd-win` tries to use the default WSL distribution, which for some reason doesn't work. Good news: you can either make the other distribution the default, or use `usbipd attach --wsl --distribution docket-desktop`. Either should work.

dorssel, 2023-12-23:
Just for my pleasure, what does `wsl --distribution docker-desktop-data -- cat
/sys/devices/platform/vhci_hcd.0/status` do (without root)?

Chris2640BE, 2023-12-23:
Now you're losing me.
How exactly can I 'make the other distribution the default' or 'use usbipd attach --wsl --distribution docker-desktop'?
For the last I assume that I must add '--bussid=<mybussid>' at the end of the command or am I seeing that wrong?

dorssel, 2023-12-23:
The default distro can be set with `wsl --set-default`, and yes, of course you need to provide the right BUSID for the second one.

Chris2640BE, 2023-12-23:
Ok, i give it a try.

Made docker-desktop the default distribution: wsl --set-default docker-desktop
Answer: de bewerking is voltooid (using a Dutch OS), so this must be OK

Next tried to attach as follows: usbipd attach --wsl --busid=<mybusID>
Answer:
usbipd: info: Using WSL distribution 'docker-desktop' to attach; the device will be available in all WSL 2 distributions.
> that seems good news, however, on a second line I get following error
usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.
After that a blank line and then a bunch of errors:
<3>WSL (33) ERROR: CreateProcessParseCommon:700: Failed to translate C:\Windows\system32
<3>WSL (33) ERROR: CreateProcessParseCommon:754: getpwwod(0) failed 2
<3>WSL (33) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Windows\system32
<3>WSL (33) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Windows
<3>WSL (33) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Windows\System32\Wbem
<3>WSL (33) ERROR: UtilTranslatePathList:2866: Failed to translate C:\Windows\System32\WindowsPowerShell\v1.0\
<3>WSL (33) ERROR: UtilTranslatePathList:2966: Failed to translate C:\Windows\System32\OpenSHH\
<3>WSL (33) ERROR: UtilTranslatePathList:2966: Failed to translate C:\Program Files\usbipd-win\
<3>WSL (33) ERROR: UtilTranslatePathList:2966: Failed to translate C:\Program Files\Docker\Docker\resources\bin
<3>WSL (33) ERROR: UtilTranslatePathList:2966: Failed to translate C:\Users\My Name\appData\Local\Microsoft\Windows\Apps
Processing fstab with mount -a failed.
Failed to mount C:\, see dmesg for more details.
Failed to mount C:\, see dmesg for more details.

Chris2640BE, 2023-12-23:
Sorry, to soon send.
Continuing errors from the one before last line:
Failed to mount C:\, see dmesg for more details.
Failed to mount D:\, see dmesg for more details.
Failed to mount C:\, see dmesg for more details.
Failed to mount D:\, see dmesg for more details.

<3>WSL (33) ERROR: CreateProcessEntryCommon:331: getpwwod(0) failed 2
<3>WSL (33) ERROR: CreateProcessEntryCommon:502: execvpe /bin/sh failed 2
<3>WSL (33) ERROR: CreateProcessEntryCommon:505: Create process not expected to return

Chris2640BE, 2023-12-23:
In the mean time I retried to install the Zigbee Home Automation (ZHA) integration in Home Assistant but the problems I experienced before are remain unchanged:
- normally, the properties of attached devices (in this case the slae.sh cc2652rb controller USB-stick) can be found in "All hardware" (Settings > System > Hardware > All hardware) but in my case the "All hardware" option doesn't show
- when I try to install the ZHA-integration I must first indicate the Radio type (ZNP in my case) and, in the following screen the "Serial device path" that normally should be found in All devices

Any idea if it is possible to find the "Serial device path" of the stick can be found somewhere else?

dorssel, 2023-12-23:
It looks like neither of the docker-desktop instances are able to perform the attach. No problem! Just install a dedicated distro, I recommend Ubuntu. Then either make that the default, or use `--distribution Ubuntu`. The device will then be available in docker as well.

Chris2640BE, 2023-12-24:
Ok, but as a Newbee I'm afraid that this exceeds by far my actual knowledge but I will give it a try.
If I understand it wel I should::
- install a new dedicated Ubuntu distribution inside the Docker Desktop container and make that the default
- copy the existing Home Assistant application that runs at this moment in the Docker-desktop/Docker-desktop-data distribution to the new Ubuntu distribution
- run the Ubuntu distribution and, if everything goes well, remove the Docker-desktop and Docker-desktop-data distributions
Please forgive me if I'm seeing this completely wrong.

Chris2640BE, 2023-12-24:
In the mean time

- I managed to install Ubuntu using the "wsl --install -d Ubuntu" command

- doing that I had to create a (new) UNIX username and a new password, which I did

- I checked the installation with the "wsl --list --all --verbose" command and found Ubuntu as a new instance beside of the docker-desktop and docker-desktop-data instances:

NAME/STATE/VERSION
(star) docker-desktop-data/Running/2
docker-desktop/Running/2
Ubuntu/Stopped/2  

- then I run the "wsl --set-default ubuntu" command in order to activate Ubuntu which gave following result:

NAME/STATE/VERSION
(star) Ubuntu/running/2
docker-desktop-data/Running/2
docker-desktop/Running/2

- next I enabled the integration with my default WSL distro by Ubuntu as additional distro via Resources>WSL Integration

So far, so good I think/hope.

Then, in a cmd-screen on the Windows host, I attached the usb-device with "usbipd attach --wsl  --busid=mybusid" and got the answer:
usbipd: info: Using WSL distribution "Ubuntu" to attach; the device will be available in all WSL 2 distributions
usbipd: info: Using IP adress xxx.xx.xxx.x to reach the host

That also looks good, I think.

Next step is to open my HA-application in Ubuntu and try to find the properties of the slae.sh cc2652rb controller USB-stick "All hardware" (Settings > System > Hardware > All hardware) in order to install the ZHA-integration.

All by all, untill now it is looking quite good and I have high hopes to bring this to a good end.

dorssel, 2023-12-26:
You did everything right installing Ubuntu. This step was not (strictly) necessary, I think:

> next I enabled the integration with my default WSL distro by Ubuntu as additional distro via Resources>WSL Integration

I think we can conclude that docker (neither its main WSL instance nor the data instance) is not capable of performing the attach. You always need a "regular" (non-docker) distro installed and running for the attach to work. I'll change the code so that it will automatically exclude the docker instances (and provide a warning). I'm not a docker user myself, so good job with all the feedback. Thanks!

Chris2640BE, 2023-12-26:
Hi Frans,
In the mean time I tried to solve the issue by installing another driver for the usb-stick on my Windows host among with some other things but all that didn't work either.
I also think that the best solution is to reinstall my system completely from scratch (once again ...) in order to make use of HAOS (Home Assistant Operating System) as this installation method seems to give the most possibilities afterwards.
I found the procedure to do this on https://www.home-assistant.io/installation/generic-x86-64, hopefully all goes well.
Before trying this I have to figure out how to make a backup of my HA-installation in order to easilly reinstalll it after the HAOS-installation. Any advice on this matter (is that überhaupt possible?) is much appreciated.
I also sincerally thank you for the effort you did in trying to resolve this issue!
Chris

dorssel, 2023-12-26:
Hi Chris,
The drivers on the Windows side (host drivers) really do not matter, as they are replaced with the VirtualBox stub driver when you attach it to WSL. If you cannot find your device in WSL after attaching, then most likely there is no *linux* driver in the default WSL kernel. That is solvable, compiling your own kernel, but it is not trivial; see https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-wsl-2-kernel-with-additional-drivers. If this machine is dedicated for your home automation, then it does make sense to use the dedicated OS like you said.

If you want to pursue the WSL option, that's possible to. Like I said, compiling the kernel is not trivial but certainly doable.

dorssel, 2023-12-29:
In the mean time, #817 handles docker-desktop better. It should no longer be necessary to install a dedicated distro.

closed 2023-12-29 by dorssel