dorssel/usbipd-win #816: some hid devices do not work immediately after attaching

comsyspro, 2023-12-27:
i know i'm here on the windows server usbipd and here all works fine without problem.

i hope getting more infos here also concerning the linux usbip server implementation. i have similar problems with some hid devices as described here:
https://github.com/dorssel/usbipd-win/issues/659#issuecomment-1677977801

after attaching the devices, they are shown in linux os in a connected state but will not work immediately. only after detaching and reattaching 1 to 3 times then the usb devices come to work correctly. so could this be a similar driver issue in linux server usbip or timeout issue?

dorssel, 2023-12-27:
Can you do some captures? See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting. Preferably one of each: a fail and a success.

comsyspro, 2023-12-27:
to understand it right, the described problem only exists when connecting to a linux server. connecting to your usbipd is working. so i asked myself why your code is working immediately and the usbip of linux is not. if i would know this i could look at the source code of usbip to check if there is a difference. do you have informations where the source code of usbip linux version is and where (in which file) probably to look into that code? i don't also know where the linux usbip project is published. so do you have here some infos?

dorssel, 2023-12-27:
Ah, I misunderstood.
Yes, this is probably timing. `usbipd-win` is user mode software, whereas Linux USBIP handles everything in kernel mode. This means there is a little more latency for each USB request when using `usbipd-win`. Maybe the device needs a little time to settle after reset/set_configuration. And Linux is "too fast". There are ways to get USB captures on Linux too, using `usbmon`. For example, see https://wiki.wireshark.org/CaptureSetup/USB, although there are ways to do it without wireshark tools from the command line as well. Try to get a fail and success capture on the *client* side.

comsyspro, 2023-12-27:
the observed hid device is a graphic tablet which has 2 bNumInterfaces, one of them is a pen to draw and the other is a touch tablet which can capture the finger touch and movement

here are reults done with usbmon capture:
9_1.mon.txt-> here the hid device was only attached one time and didn't work but was seen by the os and the tablet driver
9_3.mon.txt -> here the hid device was attached and detached 3 times in total and begin to work after these loops, the number of trys can vary, sometimes it needs 5 trys, sometimes only the pen works somteimes both the pen and the finger touch works (it's very randomly)

[9_1.mon.txt](https://github.com/dorssel/usbipd-win/files/13779160/9_1.mon.txt)

[9_3.mon.txt](https://github.com/dorssel/usbipd-win/files/13779162/9_3.mon.txt)

i hope you can dechiffre these files as i don't understand this output.

dorssel, 2023-12-28:
I've converted the traces using https://github.com/johnkeeping/usbmon2pcap:

[usbmon2pcap.zip](https://github.com/dorssel/usbipd-win/files/13788032/usbmon2pcap.zip)

Now they can be opened with wireshark. They are identical, except that 9_1 simply stops at line 110. No errors, just no further USB traffic. 9_3 happily continues. Looks like some kind of deadlock for 9_1. Maybe a race condition in the driver, something waiting forever for something that will never come... I cannot help you further, sorry.

comsyspro, 2023-12-28:
thank you for your investigation and support.

closed 2023-12-29 by dorssel