dorssel/usbipd-win #818: usbipd: error: Unable to run 'usbip' client tool.

sashanke, 2023-12-28:
Hey, i'm getting following error when i try to attach my YubiKey: "**usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.**"

```
usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-1    046d:0943  Brio 500, USB-Eingabegerät                                    Not shared
1-3    1050:0407  USB-Eingabegerät, Microsoft Usbccid-Smartcard-Leser (WUDF)    Shared
1-6    046d:0a87  Logitech G935/G933s Gaming Headset, USB-Eingabegerät          Not shared
1-14   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared

Persisted:
GUID                                  DEVICE
fe22e2cd-27bd-4531-af02-bd78bdc7ef3a  USB-Eingabegerät, Microsoft Usbccid-Smartcard-Leser (WUDF)
```

```
usbipd attach --wsl -a --busid 1-3
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.
```

```
usbipd --version
4.0.0+182.Branch.master.Sha.2ffe37ec799b9e73eb9d23d051d980fefb616ce1
```

```
wsl --version
WSL-Version: 2.0.15.0
Kernelversion: 5.15.133.1-1
WSLg-Version: 1.0.59
MSRDC-Version: 1.2.4677
Direct3D-Version: 1.611.1-81528511
DXCore-Version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows-Version: 10.0.22631.2861
```

sashanke, 2023-12-29:
After an reinstall of Ubuntu it works again. Not sure what happened.

dorssel, 2023-12-29:
Normally, I use a message like this to investigate. But since you have resolved it and it is no longer reproduceable, let's call it a fluke.

closed 2023-12-29 by dorssel