dorssel/usbipd-win #824: can not attach to wsl

heartacker, 2024-01-08:
- Windows
```log
$ usbipd attach -w  -a -b 1-10
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.
```
- wsl
```shell
acker@Acker-Liu-PC:pts/0->/home/acker (2)
> usbip
WARNING: usbip not found for kernel 5.15.133.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.15.133.1-microsoft-standard-WSL2
    linux-cloud-tools-5.15.133.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2
acker@Acker-Liu-PC:pts/0->/home/acker (2)
>
```

dorssel, 2024-01-08:
This is a known issue, which has been fixed in the master branch (but not released yet). Please try the latest master build at https://github.com/dorssel/usbipd-win/actions/runs/7380764023.

heartacker, 2024-01-09:
still get error
```
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using client tools located at /mnt/c/Program Files/usbipd-win/wsl.
usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.
```

dorssel, 2024-01-09:
What is the output of

```
"/mnt/c/Program Files/usbipd-win/wsl/usbip" version
```

in WSL?

heartacker, 2024-01-12:
I change usbip  mod to 777 and usbip works now

dorssel, 2024-01-13:
Ah, that's very interesting!
How was your C: drive mounted? Did you change anything in WSL configuration? Was it as simple as `chmod 777 /mnt/c/Program Files/usbipd-win/wsl/usbip`?

heartacker, 2024-01-13:
> Ah, that's very interesting!
> How was your C: drive mounted? Did you change anything in WSL configuration? Was it as simple as 

> `chmod 777 /mnt/c/Program Files/usbipd-win/wsl/usbip`?

yes

 I change the mount_config

mathdu, 2024-01-25:
> This is a known issue, which has been fixed in the master branch (but not released yet). Please try the latest master build at https://github.com/dorssel/usbipd-win/actions/runs/7380764023.

This worked for me! Thanks for having MSI artifacts on your merges :D

dorssel, 2024-02-04:
This has been resolved in 4.1.0

closed 2024-02-04 by dorssel