dorssel/usbipd-win #827: Not respecting [Console]::OutputEncoding in Powershell

cmj2002, 2024-01-14:
PowerShell relies on the [`System.Diagnostics.Process`](https://learn.microsoft.com/en-us/dotnet/api/system.diagnostics.process?view=net-6.0) to parse the output of applications to the pipeline. By default, it uses `[Console]::OutputEncoding` as the encoding. However, it seems that usbipd does not respect this setting and always uses UTF8 encoding for output, resulting in garbled or incorrectly parsed output when using commands like `usbipd state | ConvertFrom-Json` on non-English systems.

While it is possible to temporarily modify `[Console]::OutputEncoding` to fix the issue, I would appreciate it if usbipd could provide an optional arg to output content encoded with `[Console]::OutputEncoding` when specified. When not specifying this arg, usbipd can remain unchanged to avoid breaking any existing scripts.

### Example

```
PS C:\Users\10643> [Console]::OutputEncoding

EncodingName      : Chinese Simplified (GB2312)
WebName           : gb2312
HeaderName        : gb2312
BodyName          : gb2312
Preamble          :
WindowsCodePage   :
IsBrowserDisplay  :
IsBrowserSave     :
IsMailNewsDisplay :
IsMailNewsSave    :
IsSingleByte      : False
EncoderFallback   : System.Text.InternalEncoderBestFitFallback
DecoderFallback   : System.Text.InternalDecoderBestFitFallback
IsReadOnly        : False
CodePage          : 936

PS C:\Users\10643> usbipd state
{
  "Devices": [
{
      "BusId": "2-4",
      "ClientIPAddress": null,
      "Description": "USB 输入设备, Microsoft Usbccid Smartcard Reader (WUDF)",
      "InstanceId": "USB\\VID_1050&PID_0406\\5&242A2F40&0&4",
      "IsForced": false,
      "PersistedGuid": "4d9b5191-bd7c-4cd3-820d-697baea11a3a",
      "StubInstanceId": null
    },
    {
      "BusId": null,
      "ClientIPAddress": null,
      "Description": "USB 输入设备, Microsoft Usbccid Smartcard Reader (WUDF)",
      "InstanceId": "USB\\VID_1050&PID_0406\\5&242A2F40&0&5",
      "IsForced": false,
      "PersistedGuid": "1f812183-788d-4f8b-9332-03e08e38bbdf",
      "StubInstanceId": null
    }
  ]
}
PS C:\Users\10643> usbipd state | ConvertFrom-Json

Devices
-------
{@{BusId=2-4; ClientIPAddress=; Description=USB 杈撳叆璁惧, Microsoft Usbccid Smartcard Reader (WUDF); InstanceId=USB…
```

The description should be the same, but the output of `usbipd state | ConvertFrom-Json` is garbled and `usbipd state` is right.

cmj2002, 2024-01-14:
Reference to [relevant PowerShell issue](https://github.com/PowerShell/PowerShell/issues/17523#issuecomment-1154271811).

dorssel, 2024-01-14:
For PowerShell, I recommend the PowerShell plugin `Get-UsbipdDevice`, see https://github.com/dorssel/usbipd-win/wiki/Automation#powershell. Does that solve the problem? This outputs srtongly typed data.

cmj2002, 2024-01-15:
Thank you, this solved my problem.

closed 2024-01-15 by cmj2002