dorssel/usbipd-win #829: Unverified Primary Signature and Checksum Mismatch on Drivers

Snausage0x45, 2024-01-16:
Both VBoxUsb.sys and VBoxUsbMon.sys have a primary cert which was initially used to sign the files in 2014 (spoofing possibly aside) by a certificate issued to Oracle from Verisign,  which has not been revoked, however there are some discrepancies with the cert and it's countersigner.

Looking at VboxUsbMon.sys it was compilied in 2023

[image]

Yet the primary cert, serial 51ca009816fdbd80f120e015ee75823e, is timestamped to 2014, fails signtool verification and was only valid until Dec, 2016

[image]

[image]

with the countersigner certificate being issued by a CA which could not be verified

[image]

The cert was issued to VirtualBox for Legacy Windows Only Timestamp Kludge 2014 by VirtualBox for Legacy Windows Only Timestamp CA which could not be found

[image]

signtool output:

[image]

After initial signing the file appears to have been modified due to a mismatch in the checksums

[image]

It was then later signed in October 2023 first by an Oracle certificate then by Microsoft Window's Hardware Compatibility cert based on signing times

The second Oracle signing cert also has an issue of an untrusted root CA being used in the cert path

[image]

The included cat files only validate the last signature applied from Microsoft 

[image]

Given tools like HookSignTool and FuckCertVerifyTimeValidity, the initial timestamp being 2015 or before, plus the other discrepancies, just for posterity's sake could you validate that: 
* This is expected,
* The reason for the hashsum mismatch,
* The process of getting the drivers from Oracle to being uploaded to the repo

dorssel, 2024-01-16:
Yes, this is all known. And not a problem, since the drivers are validly signed with a Microsoft attestation signature (i.e., Windows 10+ uses the sha256 signature and allows for additional (even non-valid) signatures), as long as the attestation signature chain is correct.
Oracle has its historical reasons (XP/Vista support, mainly), which `usbipd-win` does not use/rely on.
What is your issue with `usbipd-win`?

Snausage0x45, 2024-01-16:
Yeah you're right looks like the issue is on the Oracle side, they're doing some interesting things. Though, the WHQL process is far from fool proof and has been abused in the past to sign malicious drivers so just the presence of the WHQL signature is not the panacea msft wishes it were. The fact there is a checksum mismatch on the first signature and that its back-dated to 2014, which is part of a way to abuse a loophole in the driver verification policy, plus the root CA issues on the second signature is at least out of the ordinary if nothing else.

closed 2024-01-16 by Snausage0x45