dorssel/usbipd-win #835: USB devices will not enumerate after being attached

koffes, 2024-01-18:
I have had a strange issue for quite some time with usbipd.
There are three USB devices which I have forwarded successfully to WSL. They all have the same VID and PID, but different bus_id and description. 
What is strange, is that if I turn off a single device, it disappears as expected. If I turn it on again one of two things may happen:
A: All the attached devices detach
B: The single device will not enumerate, and will thus not appear in Windows. Power cycling all devices at the same time solves this.

Any tips would be greatly appreciated.

dorssel, 2024-01-18:
Can you connect each device on its own (one at a time), and post the output of `usbipd state` for each of them?

koffes, 2024-01-18:
Attached 7-2:
```
{
      "BusId": "7-2",
      "ClientIPAddress": "172.24.192.52",
      "Description": "JLink CDC UART Port (COM35), JLink CDC UART Port (COM31), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050114502",
      "IsForced": false,
      "PersistedGuid": "620917ea-73c4-4365-8604-2c415a29a977",
      "StubInstanceId": "USB\\Vid_80EE&Pid_CAFE\\001050114502"
    },
    {
      "BusId": "9-2",
      "ClientIPAddress": null,
      "Description": "JLink CDC UART Port (COM7), JLink CDC UART Port (COM5), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050139568",
      "IsForced": false,
      "PersistedGuid": null,
      "StubInstanceId": null
    },
    {
      "BusId": "8-2",
      "ClientIPAddress": null,
      "Description": "JLink CDC UART Port (COM3), JLink CDC UART Port (COM4), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050148153",
      "IsForced": false,
      "PersistedGuid": null,
      "StubInstanceId": null
    },
```

Attached 8-2:
  ```
  {
      "BusId": "7-2",
      "ClientIPAddress": "172.24.192.52",
      "Description": "JLink CDC UART Port (COM35), JLink CDC UART Port (COM31), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050114502",
      "IsForced": false,
      "PersistedGuid": "620917ea-73c4-4365-8604-2c415a29a977",
      "StubInstanceId": "USB\\Vid_80EE&Pid_CAFE\\001050114502"
    },
    {
      "BusId": "9-2",
      "ClientIPAddress": null,
      "Description": "JLink CDC UART Port (COM7), JLink CDC UART Port (COM5), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050139568",
      "IsForced": false,
      "PersistedGuid": null,
      "StubInstanceId": null
    },
    {
      "BusId": "8-2",
      "ClientIPAddress": "172.24.192.52",
      "Description": "JLink CDC UART Port (COM3), JLink CDC UART Port (COM4), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050148153",
      "IsForced": false,
      "PersistedGuid": "7c1a2c27-692a-43bc-893f-3ed8155642e4",
      "StubInstanceId": "USB\\Vid_80EE&Pid_CAFE\\001050148153"
    },
```
 Attached 9-2:
 
 ```
 {
      "BusId": "7-2",
      "ClientIPAddress": "172.24.192.52",
      "Description": "JLink CDC UART Port (COM35), JLink CDC UART Port (COM31), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050114502",
      "IsForced": false,
      "PersistedGuid": "620917ea-73c4-4365-8604-2c415a29a977",
      "StubInstanceId": "USB\\Vid_80EE&Pid_CAFE\\001050114502"
    },
    {
      "BusId": "9-2",
      "ClientIPAddress": "172.24.192.52",
      "Description": "JLink CDC UART Port (COM7), JLink CDC UART Port (COM5), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050139568",
      "IsForced": false,
      "PersistedGuid": "1bc13d90-40d2-4436-9f2b-ba6918f6159c",
      "StubInstanceId": "USB\\Vid_80EE&Pid_CAFE\\001050139568"
    },
    {
      "BusId": "8-2",
      "ClientIPAddress": "172.24.192.52",
      "Description": "JLink CDC UART Port (COM3), JLink CDC UART Port (COM4), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050148153",
      "IsForced": false,
      "PersistedGuid": "7c1a2c27-692a-43bc-893f-3ed8155642e4",
      "StubInstanceId": "USB\\Vid_80EE&Pid_CAFE\\001050148153"
    },
```

koffes, 2024-01-18:
After that I power cycle 9-2 and it does not appear again in usbipd list. The status after power cycling 9-2:
```
    {
      "BusId": "7-2",
      "ClientIPAddress": "172.24.192.52",
      "Description": "JLink CDC UART Port (COM35), JLink CDC UART Port (COM31), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050114502",
      "IsForced": false,
      "PersistedGuid": "620917ea-73c4-4365-8604-2c415a29a977",
      "StubInstanceId": "USB\\Vid_80EE&Pid_CAFE\\001050114502"
    },
    {
      "BusId": null,
      "ClientIPAddress": null,
      "Description": "JLink CDC UART Port (COM7), JLink CDC UART Port (COM5), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050139568",
      "IsForced": false,
      "PersistedGuid": "1bc13d90-40d2-4436-9f2b-ba6918f6159c",
      "StubInstanceId": null
    },
    {
      "BusId": "8-2",
      "ClientIPAddress": "172.24.192.52",
      "Description": "JLink CDC UART Port (COM3), JLink CDC UART Port (COM4), J-Link driver, USB Mass Storage Device",
      "InstanceId": "USB\\VID_1366&PID_1051\\001050148153",
      "IsForced": false,
      "PersistedGuid": "7c1a2c27-692a-43bc-893f-3ed8155642e4",
      "StubInstanceId": "USB\\Vid_80EE&Pid_CAFE\\001050148153"
    },
```

dorssel, 2024-01-18:
If you get into the problem situation (2 devices still attached, one power cycled no longer listing). What is the status of the devices in Windows device manager?

dorssel, 2024-01-18:
As an additional test: could you bind all three of them with `--force`. Does the problem still happen in that case?

koffes, 2024-01-19:
Thanks for the fast response!
When the issue occurs, the devices are not shown in device manager. It is almost like usbipd "takes" the device,  but does not show it in `usbipd list` nor is able to forward it.

For your second point `--force` seems to help somewhat. If I force all three devices, then trigger the issue, the device will not appear in windows, but it will appear in `usbipd list`. Then I can attach it and it works.

dorssel, 2024-01-19:
> the device will not appear in windows, but it will appear in usbipd list

That is not possible. I recommend that you switch to "Devices by connection" in Device Manager. When the device is attached, the device type and driver are changed, and they may show up under a different node in "Devices by type" view (the default). Please repeat the test (also without `--force`). I suspect that the device is *really* gone from Windows, i.e. that the device is in error state.

`--force` should fix all that.

koffes, 2024-01-22:
You are indeed correct. I will close this issue. IMO, the one thing which would be good to improve, is that the device appears dead but one must run `unbind -a` and then `--force`. Don't know if there are any elegant ways around this.

closed 2024-01-22 by koffes