dorssel/usbipd-win #838: My xbox 360 controller is not working anymore after unbind

vitoo, 2024-01-22:
Hello,

i tried to redirect my xbox controller from a windows (usbipd-win as server ) to windows  ([usbip-win2](https://github.com/vadimgrn/usbip-win2) as client)

it worked one time, but after unbinding the controller from the server, and unplug/replug, it's not recognized anymore by windows 

is there a device i should delete in the device manage ? 
but nothing appear as my xbox controller so i don'tk now what to do 

Thanks for your help

closed 2024-01-22 by vitoo

Katashynskyi, 2024-02-02:
hi @vitoo, I'd like to ask, your controller displays in wsl2 in some local file explorer like nautilus as a device?

vitoo, 2024-02-02:
i don't have wsl, but on debian (usb over ip, or local usb) i can see it with the command : `lsusb` or  `dmesg`