dorssel/usbipd-win #839: nrf52840/nice!nano not shown in WSL

mangkoran, 2024-01-23:
I was trying to connect [nice!nano](https://nicekeyboards.com/docs/nice-nano/) in WSL2 VM but I couldn't find the device in `dmesg` log (unlike https://github.com/dorssel/usbipd-win/issues/755). The device seems not unplug/resets as it still stay shared.

`usbipd list`
```
BUSID  VID:PID    DEVICE                                                        STATE
7-2    239a:00b3  USB Serial Device (COM11), USB Mass Storage Device            Shared
```

Software version
```
usbipd 4.0.0+182.Branch.master.Sha.2ffe37ec799b9e73eb9d23d051d980fefb616ce1
WSL version: 2.0.14.0
Kernel version: 5.15.133.1-1
WSLg version: 1.0.59
MSRDC version: 1.2.4677
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows version: 10.0.19044.3930
```

dorssel, 2024-01-24:
What is the error when you try to attach?

mangkoran, 2024-01-24:
Ah sorry my bad I mistook "bind" and "attach". I ran `usbipd bind` only before. I tried `usbipd attach` just now and it seems it's working with the default kernel.

`usbipd list`
```
7-2    239a:00b3  USB Serial Device (COM11), USB Mass Storage Device            Attached
```

```
❯ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 239a:00b3 Adafruit nice!nano
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

Sorry for the inconvenience caused.

closed 2024-01-24 by mangkoran