dorssel/usbipd-win #840: error: Unable to run 'usbip' client tool.

lmatter, 2024-01-26:
From an administrative powershell prompt, I got this error:
> usbipd.exe attach  -b 1-4 -w                                                                    
usbipd: info: Using WSL distribution 'Ubuntu-18.04' to attach; the device will be available in all WSL 2 distributions. 
usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.  

FYI - I'm m trying to follow this guide:  https://github.com/espressif/vscode-esp-idf-extension/blob/HEAD/docs/tutorial/wsl.md

dorssel, 2024-01-26:
This has been fixed in the master branch, but not released yet. Please use the installer from the master build: https://github.com/dorssel/usbipd-win/actions/runs/7596302467

lmatter, 2024-01-26:
@dorssel Thanks!  Works great.

closed 2024-01-26 by lmatter