dorssel/usbipd-win #842: Behavior of -b option

rhoog-ine, 2024-01-29:
Using the -b option, one would think that the USB port can be shared with WSL over usb-ip. This is not the case, it is just sharing whatever device happened to be connected to the indicated USB port at the time of command invocation. Connecting another device to the same port will not automatically share and attach it. The -a option would be super useful in combo with -b if it would work that way.

Would it be possible to make the -b option work more like I describe it above?

dorssel, 2024-01-29:
Unfortunately, no. The underlying VBoxUSBMon driver requires filtering on device. It cannot filter on just busid. Therefore, `usbipd-win` has to translate the busid to other device characteristics (including VID, PID, serial number, etc.) and then exactly *that* device is going to be shared/attached.

On the other hand, it would be perfectly possible to create a (separate) service that just monitors the Windows USB ports, and automagically calls bind and/or attach whenever a new device is detectd. This is outside the scope of low-level `usbipd-win`.

In fact, auto-attach does approximately that, except that each device that you will connect to the specific busid must be bound once first (which BTW requires privileges). Once you have your devices bound (at that port, once), auto-attach will attach whatever device is connected to that port. You can even unplug one and insert another, and auto-attach will just work.

rhoog-ine, 2024-01-29:
It's a pity that you consider such monitoring service out of scope. It would be such a cool feature if you could have a USB hub connected to one USB port and all devices plugged in the outputs of the HUB would automagically be 'owned' be WSL and show up in lsusb. I see that these 'bindings' are sort of persistent now. That means that once a device was plugged into a port that triggered a 'bind', it would also be bound if plugged into another port. Windows does its best to pretend in software that USB is a 'bus', which it actually isn't. The monitoring service should therefore automatically unbind on unplug event.

dorssel, 2024-01-29:
> It would be such a cool feature if you could have a USB hub connected to one USB port

*That* isn't going to happen anyway. First of all you cannot share a hub (hubs are special in USB land, they are not simply devices themselves). And the numbering of ports behind a hub is not predictable (if they are even stable).

This is a duplicate of https://github.com/dorssel/usbipd-win/issues/747

closed 2024-01-29 by dorssel