dorssel/usbipd-win #843: mtp-probe: bus: 3, device: 29 was not an MTP device

Mark6222, 2024-01-29:
When i run sudo usbipd attach -r 10.1.216.67 -b 2-5 i get that error in syslog and it connects to the ios device/iphone, for like 2 seconds then disconnects, why is this, works fine with android devices but not with ios.

dorssel, 2024-01-29:
Quick google: https://unix.stackexchange.com/questions/26842/mounting-usb-drive-that-is-not-recognized

I don't think this is `usbipd-win` related at all. More like a `udev` configuration issue, or perhaps missing drivers on the Linux side.

Mark6222, 2024-01-31:
i tried changing my udev rules and now im getting this 
libusbip: error: udev_device_new_from_subsystem_sysname failed
usbip: error: open vhci_driver
and ive changed the udev rules back, and am still getting the same error

Mark6222, 2024-01-31:
it dosent seem to be a udev issue 
`pi@usbipserver:/etc/udev/rules.d $ udevadm monitor
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[1493.130508] add      /devices/platform/vhci_hcd.0/usb3/3-1 (usb)
KERNEL[1493.154998] add      /devices/platform/vhci_hcd.0/usb3/3-1/3-1:1.0 (usb)
KERNEL[1493.155111] bind     /devices/platform/vhci_hcd.0/usb3/3-1 (usb)
KERNEL[1493.155246] remove   /devices/platform/vhci_hcd.0/usb3/3-1/3-1:1.0 (usb)
UDEV  [1493.178027] add      /devices/platform/vhci_hcd.0/usb3/3-1 (usb)
UDEV  [1493.181412] add      /devices/platform/vhci_hcd.0/usb3/3-1/3-1:1.0 (usb)
UDEV  [1493.191857] bind     /devices/platform/vhci_hcd.0/usb3/3-1 (usb)
UDEV  [1493.193343] remove   /devices/platform/vhci_hcd.0/usb3/3-1/3-1:1.0 (usb)
KERNEL[1493.241707] unbind   /devices/platform/vhci_hcd.0/usb3/3-1 (usb)
KERNEL[1493.241794] remove   /devices/platform/vhci_hcd.0/usb3/3-1 (usb)
UDEV  [1493.244787] unbind   /devices/platform/vhci_hcd.0/usb3/3-1 (usb)
UDEV  [1493.255719] remove   /devices/platform/vhci_hcd.0/usb3/3-1 (usb)`

Mark6222, 2024-02-08:
Hi sorry been a while since i tried to fix this again, so i have the libmobiledevice library and it has usbmuxd, now for the libmobiledevice library to work usbmuxd has to be running but when it is its interfering with usbip somehow and disconnects the device.

Mark6222, 2024-02-08:
im completely lost on how to fix this

dorssel, 2024-02-08:
Please run the server from the console (as in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting) and make a PCap capture as explained in that same document. Then do an attach as before, and wait a few seconds after the error (auto detach). Then quit the server (Ctrl+C) and post the capture file.

dorssel, 2024-02-08:
That is immediately after SET_CONFIGURATION. I think this device "resets" when its configuration is changed/set. The problem is that to Windows, that looks like an unplug of the device, followed by an immediate replug. Unfortunately, the unplug already leads to a disconnect of the USBIP link. This is by design (of the USBIP protocol), and cannot be changed. I'm afraid this device is incompatible with USBIP.

Mark62222, 2024-04-10:
i fixed this issue with changing the versions of usbmuxd that i was using

closed 2024-04-26 by dorssel