dorssel/usbipd-win #846: usbipd: warning: USB filter 'USBPcap' is known to be incompatible -- what is the exact symptom of incompatibility?

rhoog-ine, 2024-01-31:
I recently installed wireshark with the usbpcap because I want to debug a usb device I'm developing. Now I get this warning in usbipd...

What is the impact of the incompatibility? Googling brings up a 'workaround' by using --force with bind (which apparently binds in a non-shared way -- host will not be able to use the device). Does that mean that the traffic for the bound device no longer shows up in wireshark?

Is the incompatibility still an issue when usbpcap isn't capturing anything at all? or is it only an issue when usbpcap is active?
The usbipd list command still works...

I think a little wiki page around this subject would be in order.

rhoog-ine, 2024-02-01:
So what you say is that I actually don't need USBPcap in the situation I already have usbipd.
USBPcap can only snoop one root hub at a time, does the trace mode of usbipd trace all hubs at once? I hope it won't result in way too much traced traffic... Does usbipd trace string descriptor requests, unlike usbpcap? I'm afraid the tracing functionality is geared for debugging usbipd, whereas usbpcap is geared for generic usb traffic tracing. Maybe that purpose makes it not trace things that a generic USB traffic analysis would be interrested in. On the other hand, usbpcap doesn't trace everything either... Linux usbmon is probably better for me.

dorssel, 2024-02-01:
`usbipd-win` only captures devices that are attached via `usbipd-win`. It captures all URBs for those devices.

closed 2024-02-12 by dorssel