dorssel/usbipd-win #851: usbipd-win 4.0.0 --WSL error

UnidentifiedTag, 2024-02-01:
v4.0.0 gives a --wsl error while v3.1.0 worked under windows 11 21h2 20000.2538

`usbipd wsl attach -d Ubuntu -b 2-2`

if i remove wsl or use --wsl with v4.0.0 it makes no change, just get an error about wsl command not being there and it being needed and then another about wsl command needing to be removed because its no longer needed!

dorssel, 2024-02-01:
What is the exact error you get with 4.0.0 and the new command syntax?

dorssel, 2024-02-04:
Besides, 4.1.0 was just released. It has improved WSL support.

UnidentifiedTag, 2024-02-04:
I haven’t tested 4.1.0 but 4.0.0 gives an —wsl command is required when the command isn’t used and when the command is used the error changes to —wsl is required, so either way you end up with a —wsl error, if you use it or not in 4.0.0, I’ll let you know in a couple of days if 4.1.0 fixes it, you have my windows build number above along with the command I used

UnidentifiedTag, 2024-02-08:
`Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Install the latest PowerShell for new features and improvements! https://aka.ms/PSWindows

PS C:\Users\signe> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-5    0c45:6a00  Integrated Webcam                                             Not shared
2-7    04f3:020b  USB Input Device                                              Not shared
3-6    8087:07dc  Intel(R) Wireless Bluetooth(R)                                Not shared

Persisted:
GUID                                  DEVICE

PS C:\Users\signe> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-2    10c4:ea60  Silicon Labs CP210x USB to UART Bridge (COM1)                 Not shared
2-5    0c45:6a00  Integrated Webcam                                             Not shared
2-7    04f3:020b  USB Input Device                                              Not shared
3-6    8087:07dc  Intel(R) Wireless Bluetooth(R)                                Not shared

Persisted:
GUID                                  DEVICE

PS C:\Users\signe> usbipd wsl attach -d Ubuntu -b 2-2
usbipd: error: The 'wsl' subcommand has been removed. Learn about the new syntax at https://learn.microsoft.com/windows/wsl/connect-usb#attach-a-usb-device.
PS C:\Users\signe> usbipd wsl attach -d Ubuntu -b 2-2
usbipd: error: The 'wsl' subcommand has been removed. Learn about the new syntax at https://learn.microsoft.com/windows/wsl/connect-usb#attach-a-usb-device.
PS C:\Users\signe> usbipd attach -d Ubuntu -b 2-2
'-d' was not matched. Did you mean one of the following?
-a
-b
-i
-w
Option '--wsl' is required.
Unrecognized command or argument '-d'.
Unrecognized command or argument 'Ubuntu'.

usbipd-win 4.1.0

Description:
  Attaches a USB device to a client.

  Currently, only WSL is supported. Other clients need to perform an attach using client-side tooling.

  Exactly one of the options '--busid' or '--hardware-id' is required.

Usage:
  usbipd attach [options]

Options:
  -a, --auto-attach                      Automatically re-attach when the device is detached or unplugged
  -b, --busid <BUSID>                    Attach device having <BUSID>
  -i, --hardware-id <VID:PID>            Attach device having <VID>:<PID>
  -w, --wsl <[DISTRIBUTION]> (REQUIRED)  Attach to WSL, optionally specifying the distribution to use
  -?, -h, --help                         Show help and usage information

PS C:\Users\signe> usbipd --wsl attach -d Ubuntu -b 2-2
'--wsl' was not matched. Did you mean one of the following?
--wsl
'-d' was not matched. Did you mean one of the following?
-a
-b
-i
-w
Option '--wsl' is required.
Unrecognized command or argument '--wsl'.
Unrecognized command or argument '-d'.
Unrecognized command or argument 'Ubuntu'.

usbipd-win 4.1.0

Description:
  Attaches a USB device to a client.

  Currently, only WSL is supported. Other clients need to perform an attach using client-side tooling.

  Exactly one of the options '--busid' or '--hardware-id' is required.

Usage:
  usbipd attach [options]

Options:
  -a, --auto-attach                      Automatically re-attach when the device is detached or unplugged
  -b, --busid <BUSID>                    Attach device having <BUSID>
  -i, --hardware-id <VID:PID>            Attach device having <VID>:<PID>
  -w, --wsl <[DISTRIBUTION]> (REQUIRED)  Attach to WSL, optionally specifying the distribution to use
  -?, -h, --help                         Show help and usage information

PS C:\Users\signe>`

dorssel, 2024-02-08:
Options come *after* the command. `--wsl` is an option of the `attach` command. Try:

```
usbipd attach --wsl -b 2-2
```

UnidentifiedTag, 2024-02-08:
the above command works on v3.1.0 while it is failing on 4.0.0 and 4.1.0, after using the new command you shared it worked, the problem is people are use to ruining the command the other way and you have made it mandatory to run this new way, this is why you are getting issues opening up about this, may i make the suggestion you allow it to be run which ever way people wish to type the command?

closed 2024-02-08 by UnidentifiedTag