dorssel/usbipd-win #856: error: Mounting 'C:\Program Files\usbipd-win\WSL' within WSL failed.

kaeltis, 2024-02-06:
Hi, since updating to 4.1.0 I can no longer attach devices to my Arch WSL instance.
When trying `usbipd attach -w -b 5-3` I get the message:
```
usbipd: info: Using WSL distribution 'Arch' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: Mounting 'C:\Program Files\usbipd-win\WSL' within WSL failed.
```

However, I can access the folder from within WSL without problems:
[image]

Since `WSL` is uppercase in the message, I tried renaming the folder, but that didn't solve the issue.

dorssel, 2024-02-06:
Since 4.1.0, there will always be a new mount point created (the one you list does not always exist, or as a different path, or non-executable).

After failure, what is the output of (in WSL):
```
ls -l /var/run/usbipd-win
```

What is the output of the following commands (as root, so possibly prepend `sudo`):
```
mkdir -m 0000 "/var/run/usbipd-win"
mount -t drvfs -o "uid=0;gid=0;umask=222" "C:\Program Files\usbipd-win\WSL" "/var/run/usbipd-win"
```

NOTE: please include all the quotes, these are the exact commands that `usbipd-win` issues (on your machine).

Hammerflame, 2024-02-06:
i have the same issue to my kali-linux WSL2 on windows pc

dorssel, 2024-02-07:
@kaeltis 
```
❯ sudo mount -t drvfs -o "uid=0;gid=0;umask=222" "C:\Program Files\usbipd-win\WSL" "/var/run/usbipd-win"
mount: /var/run/usbipd-win: failed to parse mount options 'rw,uid=0;gid=0;umask=222': Invalid argument.
```
Note how the error included "rw," and separates with a comma, where the other options are separated with a semicolon.

Also, it turns out that we do not need uid/gid, and mounting with "ro" is actually better.

Could you try:

```
sudo mount -t drvfs -o "ro,umask=222" "C:\Program Files\usbipd-win\WSL" "/var/run/usbipd-win"
```

This also works on Ubuntu (which appears to accept both semicolon and comma). Hopefully we'll find a syntax that works everywhere...

kaeltis, 2024-02-07:
> Could you try:
> 
> ```
> sudo mount -t drvfs -o "ro,umask=222" "C:\Program Files\usbipd-win\WSL" "/var/run/usbipd-win"
> ```

This works, and while the umask option doesn't show up under `mount`, the permissions of the files look good:
```
drvfs on /run/usbipd-win type 9p (ro,relatime,dirsync,aname=drvfs;path=C:\Program)

❯ ls -alh /var/run/usbipd-win/
Permissions Size User Date Modified Name
.r-xr-xr-x  3.3k root  2 Feb 17:32   auto-attach.sh
.r--r--r--  2.9k root  2 Feb 17:32   README.md
.r-xr-xr-x  1.2M root  2 Feb 17:32   usbip
```

Weirdly, the automatic mounts of the Windows drives use `uid=1000;gid=1000` - even with the semicolon:
```
drvfs on /mnt/c type 9p (rw,noatime,dirsync,aname=drvfs;path=C:\;uid=1000;gid=1000;symlinkroot=/mnt/,mmap,access=client,msize=262144,trans=virtio)
```

dorssel, 2024-02-07:
@kaeltis @Hammerflame 
Could you try the (dev) installer from https://github.com/dorssel/usbipd-win/actions/runs/7813976713
Please, shutdown WSL first (with `wsl --shutdown`), to remove any left-over mounts. And then test if `usbipd attach` works out-of-the-box.

kaeltis, 2024-02-07:
With the dev installer everything is working again and the mount is there, thanks a bunch!

```
drvfsa /run/usbipd-win 9p ro,dirsync,relatime,aname=drvfs;path=C:\Program Files\usbipd-win\WSL;umask=222;symlinkroot=/mnt/,mmap,access=client,msize=262144,trans=virtio 0
```

closed 2024-02-07 by dorssel

proddy, 2024-02-10:
4.1.1 worked for me too (same issue). thanks

enema-combatant, 2024-02-20:
Can confirm:

1. Shutdown WSL
2. Uninstall Official MSI
3. Install Dev Build (not signed, will warn, SmartScreen triggers also)
4. Fire up WSL
5. Admin Command Prompt - bind device
6. Normal Command Prompt - attach device
7. lsusb in ubuntu shows device

jrozner, 2024-03-09:
You planning to cut a 4.1.1 release soon? I'm running into this too

dorssel, 2024-03-09:
@jrozner 
Just install the latest master build until then. https://github.com/dorssel/usbipd-win/actions/workflows/build-installer.yml?query=branch%3Amaster

cmonty14, 2024-03-14:
I installed the lastest master:
```
usbipd --version                                                                                        4.1.1+26.Branch.master.Sha.ef7dda876d6d2a579fc7a6468990c3ac5d19ea13
```
And performed the steps instructed [here](https://github.com/dorssel/usbipd-win/issues/856#issuecomment-1953344712).

However, I still cannot attach an USB device; here's the relevant output:
```
usbipd list                                                                                                                                                                            ─╯
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-1    0781:5530  USB-Massenspeichergerät                                       Shared (forced)
2-3    06cb:00bd  Synaptics UWP WBDI                                            Not shared
2-4    13d3:5406  Integrated Camera, Integrated IR Camera, Camera DFU Device    Not shared
2-10   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared
3-6    0bda:8153  Realtek(R) USB GbE Family Controller                          Not shared
4-3    046d:c548  Logitech USB Input Device, USB-Eingabegerät                   Not shared
4-7    0424:284c  Hub Feature Controller, USB-Eingabegerät                      Not shared
5-1    20a0:4108  USB-Eingabegerät, Microsoft Usbccid-Smartcard-Leser (WUDF)    Not shared
5-2    0b0e:245e  Jabra Link 370, USB-Eingabegerät                              Not shared
5-4    046d:c52b  Logitech USB Input Device, USB-Eingabegerät                   Not shared

Persisted:
GUID                                  DEVICE

usbipd attach --wsl --busid=2-1                                                                                                                                                     
usbipd: info: Using WSL distribution 'Arch' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.17.160.1 to reach the host.
WSL usbip: error: Attach Request for 2-1 failed - Device busy (exported)
usbipd: warning: The device appears to be used by Windows; stop the software using the device, or bind the device using the '--force' option.
usbipd: error: Failed to attach device with busid '2-1'.
```

dorssel, 2024-03-14:
@cmonty14 
See https://github.com/dorssel/usbipd-win/issues/447#issuecomment-1284155716

cmonty14, 2024-03-14:
> @cmonty14 See [#447 (comment)](https://github.com/dorssel/usbipd-win/issues/447#issuecomment-1284155716)

I already used `--forced`, please check output of `usbipd list`

dorssel, 2024-03-14:
--force sometimes requires a reboot

cmonty14, 2024-03-14:
I see.
But if I reboot Windows 11, then usbipd is going down and must be restarted.

dorssel, 2024-03-14:
--force is permanent and survives reboot.

quispiam, 2024-06-04:
FYI i had the same error `usbipd: error: Mounting 'C:\Program Files\usbipd-win\WSL' within WSL failed.` while trying to access usb devices in NixOS running in WSL2 (https://github.com/nix-community/NixOS-WSL).

When i ran 
`ls -l /var/run/usbipd-win`
after running the failed usbpid attach there was `No such file or directory`

But your recommendation of running the following from within WSL
`mkdir -m 0000 "/var/run/usbipd-win"`
followed by
> Could you try:
> 
> ```
> sudo mount -t drvfs -o "ro,umask=222" "C:\Program Files\usbipd-win\WSL" "/var/run/usbipd-win"
> ```
> 
> This also works on Ubuntu (which appears to accept both semicolon and comma). Hopefully we'll find a syntax that works everywhere...

worked fine.

I tried downloaded the latest master and installing it 

```
usbipd --version
4.2.1+37.Branch.master.Sha.514abfa49b485f677444a6d138b47b1a6047a064
```

but that didn't seem to work, it still threw the same error `usbipd: error: Mounting 'C:\Program Files\usbipd-win\WSL' within WSL failed.`

In the end i just added the two commands to create and mount the directory to my nixos configuration.nix as follows:
```
{ config, lib, pkgs, inputs, ... }:

let
  # This is needed because usbipd for windows fails to create these
  # directories and mount them with it's inbuilt commands
  usbipd-mount-sh = ''
    mkdir -m 0000 "/var/run/usbipd-win"
    mount -t drvfs -o "ro,umask=222" "C:\Program Files\usbipd-win\WSL" "/var/run/usbip>  
'';

in
{
  wsl = {
    enable = true;
    defaultUser = "nixos";
    nativeSystemd = true;
    usbip.enable = true;
  };

  system.activationScripts.usbipdMount = {
    text = usbipd-mount-sh;
  };
  
  # all your other nix config stuff here
  
 }
```

dorssel, 2024-06-06:
@quispiam 
Are you saying the directory creation already doesn't work correctly? What is your default shell (`/bin/sh`)?

quispiam, 2024-06-13:
@dorssel yeah, I had exactly the same error as the OP:

> When trying usbipd attach -w -b 5-3 I get the message:
> 
> usbipd: info: Using WSL distribution 'Arch' to attach; the device will be available in all WSL 2 distributions.
> usbipd: error: Mounting 'C:\Program Files\usbipd-win\WSL' within WSL failed.

Then i found this github issue. I check the directory and it wasn't being created, so i created it manually from within my Nixos WSL2 instance and then it worked fine.

There may be something weird about Nixos - i'm very new to it but people say it does do some things different to more standard linux distros like ubuntu.

My default shell is:

> [nixos@nixos:/]$ /bin/sh --version
> GNU bash, version 5.2.26(1)-release (x86_64-pc-linux-gnu)
> Copyright (C) 2022 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> 
> This is free software; you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.

dorssel, 2024-06-14:
@quispiam 
See #973 for other NixOS issues (and a hint to just use another distro simply for attaching).

PR #859 (2024-02-07): Fix mount options
Fixes #856