dorssel/usbipd-win #861: usbipd: error: WSL kernel is not USBIP capable

Lotharyx, 2024-02-07:
I'm sure I'm missing something obvious.  Let me know what other info I need to provide.  I've snipped out some verbosity of the following Powershell transcript.  I'm on Windows 10 Version 21H2 (Build 19044.2006).

```
PS> wsl -l -v
  NAME      STATE           VERSION
* Debian    Running         2
PS> usbipd list
// etc...
13-3    1366:1015  JLink CDC UART Port (COM6), J-Link driver, USB Mass Stora...  Not shared
// etc...
PS> usbipd bind --busid 13-3
PS> usbipd attach --wsl --busid 13-3
usbipd: info: Using WSL distribution 'Debian' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: WSL kernel is not USBIP capable; update with 'wsl --update'.
PS> wsl --update
Checking for updates...
No updates are available.
Kernel version: 5.10.102.1
```
and the output of `uname -a` in the WSL environment agrees:
`Linux xxxxxxxxx 5.10.102.1-microsoft-standard-WSL2 #1 SMP Wed Mar 2 00:30:59 UTC 2022 x86_64 GNU/Linux`

Am I simply using too old a Windows version for this to work?  The [related guide](https://learn.microsoft.com/en-us/windows/wsl/connect-usb) only says "*(Windows 10 support is possible, see note below)*" without indicating what specific version is required.

Thanks!

dorssel, 2024-02-07:
This *should* work fine, in the sense that Windows 10 and the kernel version are good.

From within your distribution, what is the output of the following command:

```
cat /sys/devices/platform/vhci_hcd.0/status
```

This is how `usbipd-win` finds out if USB/IP is supported, and if it fails, you get the error you got.

Lotharyx, 2024-02-07:
Thanks for the engagement!

```
$ cat /sys/devices/platform/vhci_hcd.0/status
hub port sta spd dev      sockfd local_busid
hs  0000 004 000 00000000 000000 0-0
hs  0001 004 000 00000000 000000 0-0
hs  0002 004 000 00000000 000000 0-0
hs  0003 004 000 00000000 000000 0-0
hs  0004 004 000 00000000 000000 0-0
hs  0005 004 000 00000000 000000 0-0
hs  0006 004 000 00000000 000000 0-0
hs  0007 004 000 00000000 000000 0-0
ss  0008 004 000 00000000 000000 0-0
ss  0009 004 000 00000000 000000 0-0
ss  0010 004 000 00000000 000000 0-0
ss  0011 004 000 00000000 000000 0-0
ss  0012 004 000 00000000 000000 0-0
ss  0013 004 000 00000000 000000 0-0
ss  0014 004 000 00000000 000000 0-0
ss  0015 004 000 00000000 000000 0-0
```

All Greek to me; hopefully that tells you something...

dorssel, 2024-02-07:
Yes, that tells me something is wrong with the detection, as this is the correct output.

While keeping your distribution running (keep a command prompt to it open), then from Windows, what is the output of:
```
wsl.exe --distribution Debian --user root --cd / --shell-type none --exec cat /sys/devices/platform/vhci_hcd.0/status
```

Lotharyx, 2024-02-08:
wsl.exe didn't like the `--shell-type` argument so I had to omit it.  Otherwise:
```
PS> wsl --distribution Debian --user root --cd / --exec cat /sys/devices/platform/vhci_hcd.0/status 
hub port sta spd dev      sockfd local_busid
hs  0000 004 000 00000000 000000 0-0
hs  0001 004 000 00000000 000000 0-0
hs  0002 004 000 00000000 000000 0-0
hs  0003 004 000 00000000 000000 0-0
hs  0004 004 000 00000000 000000 0-0
hs  0005 004 000 00000000 000000 0-0
hs  0006 004 000 00000000 000000 0-0
hs  0007 004 000 00000000 000000 0-0
ss  0008 004 000 00000000 000000 0-0
ss  0009 004 000 00000000 000000 0-0
ss  0010 004 000 00000000 000000 0-0
ss  0011 004 000 00000000 000000 0-0
ss  0012 004 000 00000000 000000 0-0
ss  0013 004 000 00000000 000000 0-0
ss  0014 004 000 00000000 000000 0-0
ss  0015 004 000 00000000 000000 0-0
```
Looks pretty similar to what comes up inside WSL.

dorssel, 2024-02-08:
Are you sure the original error wasn't a fluke? Maybe the WSL distro was in the process of starting up or being shut down?
Can you reconfirm that the following command still gives that error?

```
usbipd --version
usbipd attach --wsl --busid 13-3
```

Lotharyx, 2024-02-08:
Yep, already tried again, same error.
```
PS > usbipd attach --wsl --busid 13-3
usbipd: info: Using WSL distribution 'Debian' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: WSL kernel is not USBIP capable; update with 'wsl --update'.
PS > usbipd --version
4.1.0+52.Branch.master.Sha.b0b7589d2dc4481b1af481787d6d773f46d0758a
```
The host has also been rebooted a couple times since starting on this journey.

Lotharyx, 2024-02-08:
```
PS > wsl --distribution Debian --user root --cd / --exec cat /sys/devices/platform/vhci_hcd.0/status 
hub port sta spd dev      sockfd local_busid
hs  0000 004 000 00000000 000000 0-0
hs  0001 004 000 00000000 000000 0-0
hs  0002 004 000 00000000 000000 0-0
hs  0003 004 000 00000000 000000 0-0
hs  0004 004 000 00000000 000000 0-0
hs  0005 004 000 00000000 000000 0-0
hs  0006 004 000 00000000 000000 0-0
hs  0007 004 000 00000000 000000 0-0
ss  0008 004 000 00000000 000000 0-0
ss  0009 004 000 00000000 000000 0-0
ss  0010 004 000 00000000 000000 0-0
ss  0011 004 000 00000000 000000 0-0
ss  0012 004 000 00000000 000000 0-0
ss  0013 004 000 00000000 000000 0-0
ss  0014 004 000 00000000 000000 0-0
ss  0015 004 000 00000000 000000 0-0
PS > $LASTEXITCODE
0
PS C:\Windows\system32> $env:WINDIR
C:\Windows
PS > dir $env:WINDIR\System32\wsl.exe

    Directory: C:\Windows\System32

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a----         9/12/2022   1:54 PM         107520 wsl.exe

PS > Get-Command wsl.exe

CommandType     Name                                               Version    Source
-----------     ----                                               -------    ------
Application     wsl.exe                                            10.0.19... C:\Windows\system32\wsl.exe
```

Again I'll point out that **wsl.exe doesn't like the --shell-type argument**.  If USBIPD is passing that in the exec, it is failing:

```
PS >  wsl --distribution Debian --user root --cd / --shell-type none --exec cat /sys/devices/platform/vhci_hcd.0/status
Invalid command line option: --shell-type
// wsl usage output...
PS > $LASTEXITCODE
-1
```

dorssel, 2024-02-08:
Oops, I missed your comment. Sorry.

> wsl.exe didn't like the --shell-type argument so I had to omit it. 

So the problem is not the distribution, or the WSL kernel, but the `wsl.exe` executable itself! Now we're getting somewhere!

Can you do in PowerShell:

```
systeminfo | findstr /B /C:"OS Name" /B /C:"OS Version"
wsl --version
wsl --help
```

Maybe (just maybe), we can just leave it out since we already use `--exec`, see https://github.com/microsoft/WSL/issues/10418. As you see from that open ticket, documentation is poor on this one...

dorssel, 2024-02-08:
I did some testing myself. Note, the goal is to *bypass* the shell and run a command directly (instead of running the shell which then interprets the command line). The reason is that we don't know what shell the user uses (probably bash, but Docker Desktop doesn't even have bash; it could even be PowerShell!); we need portability.

```
PS> wsl non-existing-command
/bin/bash: line 1: non-existing-command: command not found
PS> wsl -- non-existing-command
/bin/bash: line 1: non-existing-command: command not found
PS> wsl --shell-type standard non-existing-command
/bin/bash: line 1: non-existing-command: command not found
PS> wsl --shell-type none non-existing-command
<3>WSL (318) ERROR: CreateProcessEntryCommon:502: execvpe non-existing-command failed 2
<3>WSL (318) ERROR: CreateProcessEntryCommon:505: Create process not expected to return
PS> wsl --exec non-existing-command
<3>WSL (327) ERROR: CreateProcessEntryCommon:502: execvpe non-existing-command failed 2
<3>WSL (327) ERROR: CreateProcessEntryCommon:505: Create process not expected to return
PS> wsl --shell-type none --exec non-existing-command
<3>WSL (336) ERROR: CreateProcessEntryCommon:502: execvpe non-existing-command failed 2
<3>WSL (336) ERROR: CreateProcessEntryCommon:505: Create process not expected to return
```

As you can see `--exec` and `--shell-type none` do the same thing. We don't need them both!

Nevertheless, I still think it is a good idea to check your `wsl.exe` version. And read the first paragraph of https://learn.microsoft.com/en-us/windows/wsl/basic-commands (the part about using the Microsoft Store version of WSL). Maybe you are running the original (and by now outdated) version that came with your Windows 10.

Anyway, I will remove the `--shell-type none`, as we don't need it. Thanks for the reports!

Lotharyx, 2024-02-08:
My wsl.exe does not recognize the --version argument.
```
PS > systeminfo | findstr /B /C:"OS Name" /B /C:"OS Version"
OS Name:                   Microsoft Windows 10 Pro
OS Version:                10.0.19044 N/A Build 19044
PS > wsl --help
Copyright (c) Microsoft Corporation. All rights reserved.

Usage: wsl.exe [Argument] [Options...] [CommandLine]

Arguments for running Linux binaries:

    If no command line is provided, wsl.exe launches the default shell.

    --exec, -e <CommandLine>
        Execute the specified command without using the default Linux shell.

    --
        Pass the remaining command line as is.

Options:
    --cd <Directory>
        Sets the specified directory as the current working directory.
        If ~ is used the Linux user's home path will be used. If the path begins
        with a / character, it will be interpreted as an absolute Linux path.
        Otherwise, the value must be an absolute Windows path.

    --distribution, -d <Distro>
        Run the specified distribution.

    --user, -u <UserName>
        Run as the specified user.

Arguments for managing Windows Subsystem for Linux:

    --help
        Display usage information.

    --install [Options]
        Install additional Windows Subsystem for Linux distributions.
        For a list of valid distributions, use 'wsl --list --online'.

        Options:
            --distribution, -d [Argument]
                Downloads and installs a distribution by name.

                Arguments:
                    A valid distribution name (not case sensitive).

                Examples:
                    wsl --install -d Ubuntu
                    wsl --install --distribution Debian

    --set-default-version <Version>
        Changes the default install version for new distributions.

    --shutdown
        Immediately terminates all running distributions and the WSL 2
        lightweight utility virtual machine.

    --status
        Show the status of Windows Subsystem for Linux.

    --update [Options]
        If no options are specified, the WSL 2 kernel will be updated
        to the latest version.

        Options:
            --rollback
                Revert to the previous version of the WSL 2 kernel.

Arguments for managing distributions in Windows Subsystem for Linux:

    --export <Distro> <FileName>
        Exports the distribution to a tar file.
        The filename can be - for standard output.

    --import <Distro> <InstallLocation> <FileName> [Options]
        Imports the specified tar file as a new distribution.
        The filename can be - for standard input.

        Options:
            --version <Version>
                Specifies the version to use for the new distribution.

    --list, -l [Options]
        Lists distributions.

        Options:
            --all
                List all distributions, including distributions that are
                currently being installed or uninstalled.

            --running
                List only distributions that are currently running.

            --quiet, -q
                Only show distribution names.

            --verbose, -v
                Show detailed information about all distributions.

            --online, -o
                Displays a list of available distributions for install with 'wsl --install'.

    --set-default, -s <Distro>
        Sets the distribution as the default.

    --set-version <Distro> <Version>
        Changes the version of the specified distribution.

    --terminate, -t <Distro>
        Terminates the specified distribution.

    --unregister <Distro>
        Unregisters the distribution and deletes the root filesystem.
```

dorssel, 2024-02-08:
Please try the dev installer from https://github.com/dorssel/usbipd-win/actions/runs/7836157578. Can you confirm it fixes your problem?

You also may want to read https://learn.microsoft.com/en-us/lifecycle/announcements/windows-10-21h2-end-of-servicing
Root cause is an older wsl.exe version, from a Windows version that is EOL and/or not switching to the recommended Store version of WSL. Nevertheless, we made it work! Thanks for hanging in there.

jeanmichelleconte, 2024-02-10:
@dorssel  I had the same issue, usbipd: error: WSL kernel is not USBIP capable;  even with the latest kernel. I tried your dev installer and that did fix it for me. I can see the USB device now in wsl.

closed 2024-02-10 by dorssel

Lotharyx, 2024-02-10:
Seems to have done the trick.  On the Windows side, the error is gone, replaced with a status message about an IP address used to reach the host, and the device is now listed in WSL when I do `lsusb`.

Possibly unrelated, but I discovered this in the kernel log in WSL immediately preceding the enumeration of the USBIPD attachment:
```
[250601.540950] init: (18532) ERROR: CreateProcessEntryCommon:621: execvpe /bin/wslinfo failed
[250601.540958] 2
[250601.541033] init: (18532) ERROR: CreateProcessEntryCommon:631: Create process not expected to return
```
The only apparent connection is that it happened just over a second before the messages from `vhci_hcd`:
```
[250602.906541] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[250602.906569] vhci_hcd vhci_hcd.0: devid(851971) speed(3) speed_str(high-speed)
[250602.906651] vhci_hcd vhci_hcd.0: Device attached
```

Anyway, I report that only for informational purposes; I agree with the closure of this issue.  Thank you for your rapid response!

dorssel, 2024-02-10:
> [250601.540950] init: (18532) ERROR: CreateProcessEntryCommon:621: execvpe /bin/wslinfo failed

That's benign. It is the detection whether "mirrored networking mode" is enabled, which exists only for Window 11. If the command fails (which it does on your older WSL), then it is simply interpreted as "not in mirrored mode" (which is correct).

lanru, 2024-06-15:
@dorssel the dev installer is expired ,can you reupload it,thanks.

dorssel, 2024-06-16:
@lanru 
Just get the latest master build; is is always available.

HrMitrev, 2024-07-18:
Hi! Today I updated to the latest wsl2 version:
```
> wsl --version
WSL version: 2.3.11.0
Kernel version: 6.6.36.3-1
WSLg version: 1.0.63
MSRDC version: 1.2.5326
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.26100.1-240331-1435.ge-release
Windows version: 10.0.22631.3880
```
and I am getting this: `usbipd: error: WSL kernel is not USBIP capable; update with 'wsl --update'.`

Is this because the kernel is very different from the previous 5.15 version or they just haven't implemented the USBIP part?

dorssel, 2024-07-18:
@HrMitrev see #995