dorssel/usbipd-win #865: Tested Devices wiki page still mentions `usbipd wsl`

polyzen, 2024-02-12:
I would like to get the Nitrokey 3 added to the list, but it seems the verification process needs a bit of updating.

dorssel, 2024-02-12:
Thanks for the report!

I forgot about that page (wasn't written by me). I fixed it and would appreciate it if you gave it a little review.

What is the VID:PID of the device you want to report as working?

polyzen, 2024-02-13:
> Thanks for the report!
> 
> I forgot about that page (wasn't written by me). I fixed it and would appreciate it if you gave it a little review.

Looks good to me. Should I copy and paste it into another issue and fill it out?

> What is the VID:PID of the device you want to report as working?

`20a0:42b2`

closed 2024-02-13 by polyzen