dorssel/usbipd-win #880: Unable to attach Xbox controller to WSL

xyang2013, 2024-03-01:
versions:
usbipd-win_4.1.0
windows 11 pro OS Build 22631.3155
WSL version: 2.0.14.0
Kernel version: 5.15.133.1-1

It lists the Xbox controller
6-1   vid:pid   Xbox One Controller                                           Shared

After running the following command:
usbipd attach --wsl --busid=6-1

The status is still "Shared"

I am not sure how to resolve this issue.

dorssel, 2024-03-01:
> After running the following command:

No errors?
Then it may be that the device is immediately detached again. What is in the WSL syslog?

xyang2013, 2024-03-01:
There was no errors. The output only said it was trying to attach. But once the command was finished, the status was still "shared".
How to check WSL syslog?

dorssel, 2024-03-10:
What is the *full* output of the `usbipd attach` command. Linux logs are distribution/setup specific. Use syslog, dmesg, journalctl; whatever it should be for your setup.

xyang2013, 2024-03-11:
Thank you. I am not able to access those logs now. I deleted the windows partition and installed Ubuntu directly

dorssel, 2024-03-11:
no longer reproduced

closed 2024-03-11 by dorssel