dorssel/usbipd-win #882: Failed to attach device after enable mirrored mode networking

wsjpla, 2024-03-05:
usbipd version: **4.1.0**

Failed to attach device after [enable mirrored mode networking](https://learn.microsoft.com/en-us/windows/wsl/networking#mirrored-mode-networking) :
```
C:\Windows\system32> usbipd attach -w --busid 3-3
usbipd: info: Using WSL distribution 'Ubuntu-20.04' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 127.0.0.1 to reach the host.
usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.
WSL usbip: error: tcp connect
usbipd: error: Failed to attach device with busid '3-3'.
```

And the output of `wsl --distribution Ubuntu-20.04 --user root -- bash -c "echo < /dev/tcp/127.0.0.1/3240"` is :
```
bash: connect: Connection refused
bash: /dev/tcp/127.0.0.1/3240: Connection refused
```
The output of `sc query usbipd` is :
```
SERVICE_NAME: usbipd
        TYPE               : 10  WIN32_OWN_PROCESS
        STATE              : 4  RUNNING
                                (STOPPABLE, NOT_PAUSABLE, ACCEPTS_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0
```

But once I switch the WSL2 networking to unmirrored mode (NAT), everything works fine without any warnings or errors.

dorssel, 2024-03-05:
What is the exact configuration in .wslconfig?

wsjpla, 2024-03-05:
```
[wsl2]
swap=16GB
networkingMode=mirrored
```

dorssel, 2024-03-05:
Normally, Windows Firewall accepts all localhost connections. Connecting to any port from WSL2 to 127.0.0.1 should be possible. Are you running any other security products, or is your computer in a domain with any enforced company policies? Is there any other software that you *can* connect to from WSL2 running on Windows?

wsjpla, 2024-03-05:
> Are you running any other security products?

There is no other security products installed in my computer. 

> is your computer in a domain with any enforced company policies?

No.

> Is there any other software that you can connect to from WSL2 running on Windows?

I tried other softwares, and found that they all failed to connect after enabling mirrored mode networking. 

It seems that there is a problem with my 'Ubuntu-20.04' distribution. 
I tried my 'Ubuntu-22.04' distribution, everything works properly under mirrored mode networking.

Thanks for your reply!

closed 2024-03-08 by wsjpla

WillooWisp, 2024-03-12:
I have the same problem with Debian 11 with mirrored networking mode and usbipd, where I get 127.0.0.1/3240: Connection refused. Reverting the mirrored mode makes it work again.

thomaslr, 2024-03-26:
got the same problem as @wsjpla 
turn off the mirrored networking and everything gets passed through
turn it back on and get the blocking 3240 message.

are there any logs I could provide that would help?