dorssel/usbipd-win #883: `lsblk` not displaying USB Mass Storage Device But Shows Up On `lsusb`

bootsareme, 2024-03-06:
I did see issue #345 and similar issues but they all had the problem where they did not recompile the kernel. I recompiled the kernel and here is exactly what I did. I first downloaded *usbipd-win* on to my Windows 11 device. I opened up a elevated powershell prompt and typed `usbipd list`. Here is what I got back:

```powershell
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-7    05ac:12a8  Apple Mobile Device USB Composite Device                      Attached
1-8    046d:c52b  Logitech USB Input Device, USB Input Device                   Not shared
1-12   1462:7c95  USB Input Device                                              Not shared
1-13   8087:0029  Intel(R) Wireless Bluetooth(R)                                Not shared
3-4    1a2c:4c5e  USB Input Device                                              Not shared
```
As you see, I already binded and attached BUSID 1-7. Now, I recompile the latest version of WSL using `sudo make menuconfig` making sure to enable the setting "USB Mass Storage" under device drivers. It compiled without error. I went to my home directory and I specified a `.wslconfig` pointing to my newly generated kernel. After all that is said and done, I go back to WSL2 and I run `lsusb` and I get:

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

When I run `lsblk`, I still get:
```
NAME MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda    8:0    0 388.5M  1 disk
sdb    8:16   0     4G  0 disk [SWAP]
sdc    8:32   0   256G  0 disk /mnt/wslg/distro
```

What has it there? I recompiled the kernel and enabled all the device drivers. I am at a point of not knowing what to do anymore.

dorssel, 2024-03-06:
What's in the WSL syslog when you attach?

dorssel, 2024-03-06:
If you are not running a syslog daemon (syslogd or journald), you can get live logging if you leave `dmesg` running.

This is really a Linux administration issue, judging from:

> Bus 001 Device 002: ID 05ac:12a8 Apple, Inc. iPhone 5/5C/5S/6/SE

`usbipd-win` is working fine.

bootsareme, 2024-03-07:
What does that mena by Linux administration issue? Do you mean it does not have to do with the USB passthrough?

bootsareme, 2024-03-07:
Could it be that Apple's iPhone MTP system causes Linux to not recognize it as a block device?

dorssel, 2024-03-08:
usbipd-win just forwards the device, it does not care about the device type. The fact that the attach worked means that usbipd-win is working. It is now up to the client to load a driver for it and send URBs to the device.

I don't know that device type it is, maybe a block device requiring a special driver, maybe some other device type. But as far as usbipd-win is concerned, it doesn't matter. You really need to check the client (WSL) side logs/device tree to see what device drivers are required. A normal starting point would be syslog, or `dmesg`, or `lsusb -v`.

bootsareme, 2024-03-08:
When I run `dmesg` the following is shown:
```
[    9.402343] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[    9.402847] vhci_hcd vhci_hcd.0: devid(65543) speed(3) speed_str(high-speed)
[    9.403420] vhci_hcd vhci_hcd.0: Device attached
[    9.750288] usb 1-1: new high-speed USB device number 2 using vhci_hcd
[    9.900292] usb 1-1: SetAddress Request (2) to port 0
[    9.986435] usb 1-1: New USB device found, idVendor=05ac, idProduct=12a8, bcdDevice=13.04
[    9.986961] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    9.987355] usb 1-1: Product: iPhone
[    9.987537] usb 1-1: Manufacturer: Apple Inc.
[    9.987820] usb 1-1: SerialNumber: 00008101000568A92EF2001E
```
It seems to be shown but it is not recognized as a block device. I want to mount it as a file system.

dorssel, 2024-03-08:
There is no storage class amongst the interface listed. Maybe you need to enable that in the device (either before you plug it in, or approve file access after you plug it in). All seems working correctly otherwise.

bootsareme, 2024-03-08:
So as this project's developer, you have no control whether that storage
class gets recognized or not, confirm?

dorssel, 2024-03-09:
The device (in its current configuration) simply does not expose a mass storage interface; the mass storage driver simply isn't activated because the *device* does not support it (in its current configuration). That has nothing to do with the kernel. You probably need to configure the device itself somehow; maybe proprietary software is needed to configure the device (it has several "Vender Specific" interfaces after all).

In the end, this has nothing to do with `usbipd-win`, which itself is entirely device type agnostic. `usbipd-win` just passes through all USB traffic irrespective of the device type.

closed 2024-03-09 by dorssel