dorssel/usbipd-win #886: TsUsbFlt incompatibility

GreatAg, 2024-03-11:
Hi, Thanks for your great open source.
I have a problem in binding and my problem is that i have to use --force options beacuse of this warning:
usbipd: warning: USB filter 'TsUsbFlt' is known to be incompatible with this software; 'bind --force' will be required.

First of all i want to know how can i resolve this?
and at the next step i want to know what is --force and what exactly do?
Thanks again.

dorssel, 2024-03-12:
> First of all i want to know how can i resolve this?

TsUsbFlt is part of Terminal Server USB forwarding. It interferes with the way the VBoxUSBMon driver (the one used by `usbipd-win`) uses dynamic binding. Either disable the TsUsbFlt (I don't know how, but probably by uninstalling/disabling some Terminal Server feature), or use `usbipd bind --force` as suggested by the warning.

> and at the next step i want to know what is --force and what exactly do?

`--force` installs the VBoxUSB stub driver permanently on the device (well: until you manually `usbipd unbind` again). This means that the device will be unusable by Windows as it is continuously claimed by `usbipd` even when the device is not attached to a client. If you don't intend to use the device on Windows anyway, this is a perfectly valid solution.

GreatAg, 2024-03-12:
Thanks for your explanations.
I resolve this problem by disabling the TsUsbFlt.
Here is the procedures:
Enter " registry editor " copy this and paste : Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{36fc9e60-c465-11cf-8056-444553540000}

and Then delete this file: UpperFilters
and then reboot the os and the you'll be fine.

[image]

You can now can close this.

dorssel, 2024-03-12:
> Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class{36fc9e60-c465-11cf-8056-444553540000}

This is exactly where the code looks to see if there are any USB filters: https://github.com/dorssel/usbipd-win/blob/master/Usbipd/ConsoleTools.cs#L243

Well done!