dorssel/usbipd-win #887: TLS support?

GreatAg, 2024-03-11:
Hi, i want to know that do you have any plan to support secure connection for this project?

GreatAg, 2024-03-11:
Thanks for your quick answer.
what about access control? it can be implemented?

dorssel, 2024-03-12:
That too isn't part of the USB/IP protocol. You can tweak the firewall rule to limit/extend the remote machines that can access the bound devices. It requires Administrator rights to bind devices in the first place, but once bound any machine that can reach port 3240 can attach any bound devices. If you want user access control, you can disable the firewall rule altogether so only local applications can attach. Using ssh tunneling you can then restrict attaching to users that have a local account.

GreatAg, 2024-03-12:
Thanks for your help, i want to use this open source in production and have some concerns about it which i share above.
Another question i have is that it is able to redirect hardware tokens? and if it has this ability can i filter somehow and just show the hardware tokens usb? and if i can do this, what parameters do you suggest for filtering hardware tokens? there is any specific class or guid exists for hardware tokens in windows?
Thanks again.

dorssel, 2024-03-12:
> is it able to redirect hardware tokens?

I wouldn't see why not. `usbipd-win` (and the USB/IP protocol in general) is device type agnostic.

> just show the hardware tokens usb?

The client will only see devices listed (via `usbip list`) that are bound *and* currently connected to the host. The administrator on the Windows host just needs to make sure only to bind hardware tokens.

> there is any specific class or guid exists for hardware tokens in windows?

No. On the host you will see all devices with `usbipd list`, and an admin needs to select which device(s) to `usbipd bind`.

dorssel, 2024-04-28:
Version 4.2.0 now has policies for better control. Closing this issue.

closed 2024-04-28 by dorssel