dorssel/usbipd-win #892: Cannot attach devices after updating to usbipd 4.1.0

dkuschmierz, 2024-03-13:
After updating to usbipd version 4.1.0 I am not able to attach any device to my WSL anymore.  While trying to attach the device I get `usbipd: error: The VBoxUsbMon driver is not correctly installed; a repair or re-install of this software should fix that`. 
Not sure what the VBoxUsbMon is exactly, but it seems like it has something with VirtualBox to do which is not installed at my PC.

dorssel, 2024-03-13:
Ah, yes, I see how that wording is confusing. I'll fix that.

What is meant is (and this probably will be the new wording):

```
usbipd: error: A driver dependency is not correctly installed; a repair or re-install of 'usbipd-win' should fix that.
```

(reason: `VBoxUsbMon` is *also* part of `usbipd-win`; VirtualBox itself is not involved. And 'this' refers to `usbipd-win` itself).

Does that make sense? (if so, I'll change the wording as suggested).

dkuschmierz, 2024-03-13:
Ah ok, then the new wording will definitely help. But anyway I got that message directly after I installed version 4.1.0 (with a restart). Shall I try to reinstall the version? For now I downgraded to 4.0.0, but could gibe it a try...

dorssel, 2024-03-13:
Apparently, something goes wrong when upgrading from 4.0.0 to 4.1.0 on your setup; I don't know why. You can always do a clean install (uninstall 4.0.0, then install 4.1.0). That *should be* without issues.

shareefj, 2024-04-05:
I've just hit this issue as well after upgrading.  The installer seemed to hang and then ask for a system restart.  Once restarting I had the same issue but an uninstall and re-install cleaned it up.  Also, no long pause in the installer this time.

closed 2024-04-06 by dorssel