dorssel/usbipd-win #898: BSOD screen when plugging USB device ( J-LINK)

aswsa-github, 2024-03-20:
Windows: W11 Business 10.2.26631 
Ubuntu version:  Ubuntu 18.04.6 LTS
usbipd: 4.1.0+52.Branch.master.Sha.b0b7589d2dc4481b1af481787d6d773f46d0758a

* Ubuntu-18.04           Running         2
  docker-desktop         Stopped         2
  docker-desktop-data    Stopped         2

Got 3 times this issue
First 2 : BSOD
* Last one : Full crash of PC, screen dead (full black) , no mouse , no keyboard , etc...
Stack trace from event viewer:

Log Name:      Application
Source:        usbipd-win
Date:          20/03/2024 10:32:52
Event ID:      3
Task Category: None
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      SMA-XXXX
Description:
Category: Usbipd.ConnectedClient
EventId: 3

An exception occurred while communicating with the client:

Exception: 
System.IO.IOException: Unable to read data from the transport connection: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond..
 ---> System.Net.Sockets.SocketException (10060): A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.
   --- End of inner exception stack trace ---
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError, CancellationToken) + 0x47
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource<System.Int32>.GetResult(Int16) + 0x58
   at System.IO.Stream.<ReadAtLeastAsyncCore>d__46.MoveNext() + 0x1bb
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Threading.Tasks.Sources.ManualResetValueTaskSourceCore`1.ThrowForFailedGetResult() + 0x13
   at System.Threading.Tasks.Sources.ManualResetValueTaskSourceCore`1.GetResult(Int16) + 0x2b
   at System.Runtime.CompilerServices.PoolingAsyncValueTaskMethodBuilder`1.StateMachineBox`1.System.Threading.Tasks.Sources.IValueTaskSource<TResult>.GetResult(Int16) + 0x20
   at Usbipd.Tools.<ReadMessageAsync>d__0.MoveNext() + 0x1b6
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at Usbipd.Interop.UsbIp.<ReadUsbIpHeaderAsync>d__17.MoveNext() + 0x10f
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at Usbipd.AttachedClient.<RunAsync>d__12.MoveNext() + 0x36b
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__7.MoveNext() + 0x132b
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__7.MoveNext() + 0xe8e
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__7.MoveNext() + 0x16c8
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at Usbipd.ConnectedClient.<RunAsync>d__5.MoveNext() + 0x300

Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="usbipd-win" />
    <EventID Qualifiers="0">3</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2024-03-20T09:32:52.5073055Z" />
    <EventRecordID>10650</EventRecordID>
    <Correlation />
    <Execution ProcessID="5400" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>SMA-XXXX</Computer>
    <Security />
  </System>
  <EventData>
    <Data>Category: Usbipd.ConnectedClient
EventId: 3

An exception occurred while communicating with the client:

Exception: 
System.IO.IOException: Unable to read data from the transport connection: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond..
 ---&gt; System.Net.Sockets.SocketException (10060): A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.
   --- End of inner exception stack trace ---
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError, CancellationToken) + 0x47
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource&lt;System.Int32&gt;.GetResult(Int16) + 0x58
   at System.IO.Stream.&lt;ReadAtLeastAsyncCore&gt;d__46.MoveNext() + 0x1bb
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Threading.Tasks.Sources.ManualResetValueTaskSourceCore`1.ThrowForFailedGetResult() + 0x13
   at System.Threading.Tasks.Sources.ManualResetValueTaskSourceCore`1.GetResult(Int16) + 0x2b
   at System.Runtime.CompilerServices.PoolingAsyncValueTaskMethodBuilder`1.StateMachineBox`1.System.Threading.Tasks.Sources.IValueTaskSource&lt;TResult&gt;.GetResult(Int16) + 0x20
   at Usbipd.Tools.&lt;ReadMessageAsync&gt;d__0.MoveNext() + 0x1b6
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at Usbipd.Interop.UsbIp.&lt;ReadUsbIpHeaderAsync&gt;d__17.MoveNext() + 0x10f
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at Usbipd.AttachedClient.&lt;RunAsync&gt;d__12.MoveNext() + 0x36b
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at Usbipd.ConnectedClient.&lt;HandleRequestImportAsync&gt;d__7.MoveNext() + 0x132b
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at Usbipd.ConnectedClient.&lt;HandleRequestImportAsync&gt;d__7.MoveNext() + 0xe8e
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at Usbipd.ConnectedClient.&lt;HandleRequestImportAsync&gt;d__7.MoveNext() + 0x16c8
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at Usbipd.ConnectedClient.&lt;RunAsync&gt;d__5.MoveNext() + 0x300
</Data>
  </EventData>
</Event>

dorssel, 2024-03-20:
`usbipd-win` is user mode code, which is not supposed to be able to crash the PC at all. I'm guessing it is actually WSL that is crashing (which has a lot of kernel mode stuff running). The exception you list is normal for a non-responsive client (timeout). That is not a crash, but normal error reporting. The service `usbipd-win` did not actually stop from that.

Did you actually have a "blue screen" (BSOD)? In that case it should display the stop-code and the kernel module causing the fault, Or just a black screen? Do you use WSLg (graphics applications in WSL)?

aswsa-github, 2024-03-20:
The last issue involved a black screen without any error message, causing the PC to completely shut down. It required a restart by holding down the power button for an extended period.

The first two issues occurred immediately after installing usbipd-win, resulting in a Blue Screen of Death (BSOD). 

Yes I am also connecting WSL2 via XFCE4 and accessing it via RDP on localhost.

aswsa-github, 2024-03-20:
The two Blue Screen of Death (BSOD) occurrences and the subsequent black screen occurred immediately after I connected a USB device that had not been previously installed into my USB hub.

aswsa-github, 2024-03-21:
#New BSOD issue today
#1- Started WSL2 , just console , no gui
#2- usbipd list
**Connected:**
**BUSID  VID:PID    DEVICE                                                        STATE**
**2-3    0bda:8153  Realtek USB GbE Family Controller                             Not shared**
**3-9    045e:0990  Surface Camera Front, Surface IR Camera Front                 Not shared**
**3-10   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared**
**4-5    2109:8886  USB Billboard Device                                          Not shared**
**7-1    0403:6001  USB Serial Converter                                          Shared**
**7-5    1366:0101  J-Link driver                                                 Not shared**
**7-6    046d:c31c  USB Input Device                                              Not shared**
**7-7    03f0:134a  USB Input Device                                              Not shared**
#3- usbipd attach --wsl --busid 7-1
** usbipd: info: Using WSL distribution 'Ubuntu-18.04' to attach; the device will be available in all WSL 2 distributions.**
** usbipd: info: Using IP address 172.29.32.1 to reach the host.**
#4- use a python script open port to see logging over seria port
** After some minutes logging on the serial port**
#5 BSOD 
** BSOD Message : Stop code IRQL_NOT_LESS_OR_EQUAL , what failed: ntoskrnl.exe

0403:6001  => FTDI usb to serial converter
Mvg

dorssel, 2024-03-21:
Thanks for the report.

Unfortunately, this is something outside of our control. Most likely it is a driver, but it doesn't say which one. It *could* be a VirtualBox driver issue (see https://github.com/dorssel/usbipd-win/issues/717), but that should have logged VBoxUSBMon.sys (or VBoxUSB.sys) as the culprit. Again, not something that can be fixed by this project. But I really think it is actually WSL itself that is crashing (given the origin `ntoskrnl,exe`). Better report this to https://github.com/microsoft/WSL/issues.