dorssel/usbipd-win #899: Exception while communicating with device

dvdfreitag, 2024-03-23:
# Version information
- USBIPD Version
    + `4.1.0+52.Branch.master.Sha.b0b7589d2dc4481b1af481787d6d773f46d0758a`
- WSL
    + `WSL version: 2.1.5.0`
    + `Kernel version: 5.15.146.1-2`
    + Distribution
        * `Ubuntu 22.04.3 LTS`
- Device information
    + WCH [CH9102](https://www.wch-ic.com/products/CH9102.html) USB to UART controller
    + USB VendorId: 0x1A86, ProductId: 0x55D4
    + I can read/write the device in Windows with Teraterm
    + Tried using both Windows Update drivers as well as latest manufacturer drivers

# Steps to reproduce

1. Stop USBIPD (in Terminal 1)
```powershell
sc stop usbipd
```
2. Start USBIPD server in terminal
```powershell
usbipd server
```
3. Bind device (in Terminal 2)
```powershell
usbipd bind -b <Bus Number>
```
4. Attempt to attach device
```powershell
usbipd attach --wsl -b <Bus Number>
```
5. Error from USBIPD
```
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address <IP> to reach the host.
WSL usbip: error: Attach Request for 6-2 failed - Device in error state
usbipd: error: Failed to attach device with busid '6-2'.
```
6. Error from USBIPD Server
```
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.ComponentModel.Win32Exception (433): DeviceIoControl returned error ERROR_NO_SUCH_DEVICE
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.VBoxUsb.<ClaimDeviceOnce>d__0.MoveNext() + 0x2de
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.VBoxUsb.<ClaimDevice>d__1.MoveNext() + 0x145
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__7.MoveNext() + 0xc2e
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__7.MoveNext() + 0x16c8
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.ConnectedClient.<RunAsync>d__5.MoveNext() + 0x300
```

dorssel, 2024-03-23:
This is a normal error report for a device that is non-responsive after a dynamic driver switch. I.e., this is not a problem with `usbipd-win`, but with the device not able to cope with dynamic binding. You probably need `usbip bind --force` for this device.

dvdfreitag, 2024-03-23:
Alas, I tried `usbipd bind --force` but it didn't work. It's interesting that USBIPD-WIN v3.2.0 works just fine but v4.1.0 doesn't. I assume the implementation details regarding dynamic binding have change fairly significantly since v3.2.0, but I wonder if it's possible to add a fallback method if the new one fails?

dorssel, 2024-03-23:
I've seen update problems from <4.00 to 4.00 before. Something to do with the way the VirtualBox drivers are installed. I recommend you uninstall/reinstall 4.1.0. That should fix it.

szucsz, 2024-04-01:
I had similar (Device in error state) error during attach, in my case `usbipd bind --force` did resolve the issue. please see below error from server.

```
PS C:\WINDOWS\system32> usbipd server
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Program Files\usbipd-win\
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) + 0x128
         at Usbipd.VBoxUsb.<ClaimDeviceOnce>d__0.MoveNext() + 0x6d
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.VBoxUsb.<ClaimDevice>d__1.MoveNext() + 0x145
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at Usbipd.VBoxUsb.<ClaimDevice>d__1.MoveNext() + 0x202
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__7.MoveNext() + 0xc2e
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at Usbipd.ConnectedClient.<HandleRequestImportAsync>d__7.MoveNext() + 0x16c8
      --- End of stack trace from previous location ---
         at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
         at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
         at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
         at Usbipd.ConnectedClient.<RunAsync>d__5.MoveNext() + 0x300
info: Usbipd.ConnectedClient[1]
      Client 172.20.229.235 claimed device at 3-6 (USB\VID_303A&PID_1001\...
```

dorssel, 2024-05-01:
stale

closed 2024-05-01 by dorssel