dorssel/usbipd-win #905: Unable to install due to nonexistent vboxusbmon not stopping

PlasmaHH, 2024-03-27:
For some reason the installer for 4.1.0 thinks there is a vboxusbmon service running, however I don't have virtualbox installed. I also cannot find any service like that in the windows services pane or the sysinternals autoruns view.

[image]

dorssel, 2024-03-27:
The VirtualBox drivers are embedded in `usbipd-win`, they are used "under water". You do not have to have VirtualBox itself installed. I've seen reports of problems upgrading from a pre-4.0.0 version to a 4.0.0+ version. The way the drivers are installed have changed. Are you perhaps upgrading? In that case I recommend uninstalling the software first, and then do a fresh install with the newest installer. That *should* work. However, I find it strange that there is a mention of "insufficient privileges", since the installer is system-wide and requires Administrator privileges to begin with. I haven't seen such an error before...

PlasmaHH, 2024-03-28:
Yes I was upgrading from 4.0.0 to 4.1.0 and uninstalling and then reinstalling worked.

This is a win10 within a corporate environment, who knows what group policies do there, but generally I am able to restart services.

Maybe it is because usually people have administrator enabled accounts and just click on a dialogue whereas here we have a special user just for that?

PlasmaHH, 2024-03-28:
Due to #856 I had to download a 4.1.1 installer and running that over the existing 4.1.0 install worked fine, so it might just be 4.0.0 -> 4.1.0

dorssel, 2024-03-28:
Yes, that all makes sense (group policies, separate admin account, change of driver installation between versions, using dev installer).

Glad you sorted it all out and thanks for providing the feedback! Issue solved?

PlasmaHH, 2024-03-28:
Yes, for me all works fine. If the above error message could suggest an uninstall instead of update it might make things easier for other people in the future.

closed 2024-03-28 by PlasmaHH