dorssel/usbipd-win #911: Attach device that Windows has no driver for

Hendi48, 2024-03-30:
I have an ancient USB flatbed scanner where, unfortunately, there's no OEM driver support for Windows 10 and higher. I wanted to try to use it from WSL2 because SANE can talk to the device.

Sharing the device works:
```
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-4    04a5:20b0  VirtualBox USB                                                Shared
```

Attaching it always fails:
```
usbipd: info: Using WSL distribution 'Debian' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.20.80.1 to reach the host.
WSL usbip: error: Attach Request for 1-4 failed - Device in error state
usbipd: error: Failed to attach device with busid '1-4'.
```

I tried attaching another device (USB mouse) and that worked fine - no error, and it showed up under `lsusb`, so it doesn't appear to be a direct issue with the VBox driver.

In Device Manager, the device is listed with a warning triangle and says that "Drivers for this device have not been installed (code 28)". I also tried making it use the generic WinUsb driver, but the next time I use usbipd on it, it automatically switches back to "VirtualBox USB".

From some issues in this repo I learned that _theoretically_ this should be a supported scenario?

I'm using Windows 10.0.19045, latest master usbipd-win and WSL kernel 5.15.146.

Hendi48, 2024-03-31:
Turns out it was a hardware problem. The scanner didn't like my Intel USB controller, with an ASMedia controller everything is working fine.

closed 2024-03-31 by Hendi48

dorssel, 2024-03-31:
Still, it may work with the Intel control as well. Since you don't use the device at all from Windows, I recommend `usbipd bind --force`.