dorssel/usbipd-win #912: timestamp in USB capture incorrect

rhermsen, 2024-03-31:
timestamp in USB capture incorrect

Is this a known issue/limitation?

I started a USB capture about 3wks ago following:
[Wiki > Troubleshooting > USB-capture ](https://github.com/dorssel/usbipd-win/wiki/Troubleshooting#usb-capture)
with: ```usbipd server Logging:LogLevel:Default=Trace "usbipd:PcapNg:Path=C:\Users\<user>\pm-monitor_20240307__1808.pcapng"```

The capture is looks to be running well, but after opening the .pcapng file the timestamps look incorrect.
Wireshark stats indicate about 2 days of data, but actually the capture contains about 23 days of data (~310k packets a day, total of "Last packet is 7335587".

```Wireshark > Statistics > Capture File Properties > 
First packet: 2024-03-07 18:08:28
Last packet: 2024-03-09 21:22:55
Elapsed: 2 days 03:13:51
```

```>usbipd --version
4.0.0+182.Branch.master.Sha.2ffe37ec799b9e73eb9d23d051d980fefb616ce1
```

| packet number | packet timestamp | actual time from data |
| ---------------- | --------------------------------- | ---------------------------------------------- | 
| 6610146 | 2024-03-09 21:22:53 | in packet day stamp: "y ": "2024" |
| 6610148 | 2024-03-09 21:22:53 | in packet day stamp: "m":"03", "d":"29", "h":" 02", "min |
| 6610150 | 2024-03-09 21:22:53 | in packet time stamp: ":"32", " sec":58"  |
| 6610161 | 2024-03-09 21:22:54 | |
| 6610162 | 2024-03-07 18:08:31 | |
| 6610164 | 2024-03-07 18:08:31 | in packet day stamp: "y ": "2024" |
| 6610166 | 2024-03-07 18:08:31 | in packet day stamp: "m":"03 ", "d":"29","h":"02","min |
| 6610168 | 2024-03-07 18:08:31 | in packet time stamp: ":"33"," sec":"03" |
| 6685884 | 2024-03-08 00:00:00 | |
| 6995460 | 2024-03-09 00:00:01 | |

The debug output shows 131 times ```dbug: Usbipd.PcapNg[1000]
      Flushing```

dorssel, 2024-04-01:
Ah, yes, of course. Good find.

One never expects a 64-bit timer to overflow, except of course if it is timing in 100 ns *and* you inadvertently multiply it by 10 million. Then it will overflow (you guessed it) every 2.13 days.

Doesn't do harm to the capture itself, though. And fixed in https://github.com/dorssel/usbipd-win/pull/913.

Thanks for the report.

rhermsen, 2024-04-01:
Thanks a lot for fixing this!