dorssel/usbipd-win #915: usbipd: error: Unable to run 'usbip' client tool.

tendollor, 2024-04-05:
From an administrative powershell prompt, I got this error:

usbipd: error: WSL kernel is not USBIP capable; update with 'wsl --update'.
so , I try to fix this  with this guide: https://github.com/dorssel/usbipd-win/issues/861

My environment is almost exactly the same as the one in the wizard, except I'm using Ubuntu instead of Debain

```powershell
PS> wsl --update
Checking for updates...
No updates are available.
Kernel version: 5.10.102.1

PS C:\WINDOWS\system32> systeminfo | findstr /B /C:"OS Name" /B /C:"OS Version"
OS Name:                   Microsoft Windows 10 ?????
OS Version:                10.0.19043 N/A Build 19043
```

I realized that this BUG has been resolved at https://github.com/dorssel/usbipd-win/actions/runs/7836157578
So I pulled the project from the master version then Apply the pull request https://github.com/dorssel/usbipd-win/pull/891
checkout to wix5 branch

Compiled and generated usbipd-win_4.1.1.msi and installed it

try " usbipd attach --wsl --busid=7-3"  again i got
 
```powershell
PS C:\Users\mk> usbipd attach --wsl --busid=7-3
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.
```

dorssel, 2024-04-06:
Please try the master branch build (the wix5 one is not stable yet). You can find those here: https://github.com/dorssel/usbipd-win/actions/workflows/build-installer.yml?query=branch%3Amaster

dorssel, 2024-04-06:
Also note that your Windows version is out of support: https://learn.microsoft.com/en-us/windows/release-health/release-information

tendollor, 2024-04-06:
I updated my OS  to win11 . 
every thing is working good.

``` 
PS C:\WINDOWS\system32> usbipd attach --wsl --busid 7-3
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.30.0.1 to reach the host.

```

in docker container

```
root@ef05878e77c5:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux 5.10.102.1-microsoft-standard-WSL2 vhci_hcd USB/IP Virtual Host Controller
Bus 001 Device 002: ID 31c0:1234 AI       Mass Storage    
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.102.1-microsoft-standard-WSL2 vhci_hcd USB/IP Virtual Host Controller
```
If we don't want to support Win10 anymore, then there's no problem.
If you want to be compatible with the version I am asking, I apologize that I cannot provide the environment anymore. I have already upgraded to Win11
So I think we should be able to close this issue now.

dorssel, 2024-04-06:
Windows 10 is supported by `usbipd-win` (in fact, I use it myself), but your particular version of Windows is not supported by Microsoft. That's all.

closed 2024-04-06 by dorssel