dorssel/usbipd-win #917: Mounting within WSL failed

arkaw2222, 2024-04-08:
Hi there! Could you help plese.. 
trying to attach sd-card to wsl, but it gives error:
usbipd: error: Mounting 'C:\Program Files\usbipd-win\WSL' within WSL failed.

WSL 2, usbipd 4.1.0, linux distribution fedora

P.s. sorry for my lang, im russian, dont know eng as good as u :(
[image]

dorssel, 2024-04-08:
This is a known problem for some distributions (apparently including fedora). It has been fixed in the master branch. Please use the installer from the latest master branch: https://github.com/dorssel/usbipd-win/actions/runs/8581245579.

arkaw2222, 2024-04-08:
Thanks a lot, but i can see there only build-installer.yml, please guide me how to install dev build, kind man :)

dorssel, 2024-04-08:
Just below it says "Artifacts" and "msi-installer". Download ... unzip ... install.

arkaw2222, 2024-04-08:
> Just below it says "Artifacts" and "msi-installer". Download ... unzip ... install.

I found elder discussion, the same problem, but i could download dev installer, now i understood im a bit blind :)
Thanks, issue solved 👍

closed 2024-04-08 by dorssel