dorssel/usbipd-win #919: connecting devices over 3 machines

Mark62222, 2024-04-10:
Hi i was just wondering if its possible to connect a device with usbipd to another device and then from that device to another, i have tried this and it doesn't seem to work but just wanted to know if it was possible.

Thanks

dorssel, 2024-04-10:
I think that should work (I see no reason why it shouldn't), but I've never tried it.

Mark62222, 2024-04-10:
so when i connect a device and then on the device i connected it to do a usbip list -l its dosent show but lsusb it does

Mark62222, 2024-04-10:
Sorry for any confusing, what you described is what I'm trying to do, but when the phone is attached to the client on that client i cannot see the device busid using usbip as the phone doesn't, which means I cant bind it again in order to connect it again to the next client machine.

dorssel, 2024-04-10:
Hmm, apparently the Linux kernel doesn't like it. I get:

```
usbip: error: bind loop detected: device: /devices/platform/vhci_hcd.0/usb1/1-1 is attached to vhci_hcd
```

So, it looks like it is not possible...

Mark62222, 2024-04-11:
Ok thanks.

dorssel, 2024-04-28:
Not an usbipd-win issue.

closed 2024-04-28 by dorssel