dorssel/usbipd-win #922: Extremely slow sends during Fastboot flash operations

LLeavesG, 2024-04-15:
I tried flashing a google pixel5 device using fastboot in wsl2 and found it to be extremely slow when sending data.
You can see Sending 'product_a' (234012 KB)  spend [433.986s] 
Thank you for your excellent work and hope to hear from you!

Checking 'product'                                 OKAY [  0.070s]
Setting current slot to 'a'                        OKAY [  0.085s]
Sending 'boot_a' (98304 KB)                        
OKAY [180.539s]
Writing 'boot_a'                                   OKAY [  0.375s]
Sending 'dtbo_a' (16384 KB)                        OKAY [ 30.463s]
Writing 'dtbo_a'                                   OKAY [  0.180s]
Sending 'vbmeta_a' (8 KB)                          OKAY [  0.140s]
Writing 'vbmeta_a'                                 OKAY [  0.079s]
Sending 'vbmeta_system_a' (4 KB)                   OKAY [  0.140s]
Writing 'vbmeta_system_a'                          OKAY [  0.079s]
Sending 'vendor_boot_a' (98304 KB)                 OKAY [183.870s]
Writing 'vendor_boot_a'                            OKAY [  0.376s]
Rebooting into fastboot                            OKAY [  0.070s]
< waiting for any device >
Sending 'super' (4 KB)                             OKAY [  0.466s]
Updating super partition                           OKAY [  0.014s]
Resizing 'product_a'                               OKAY [  0.006s]
Resizing 'system_a'                                OKAY [  0.006s]
Resizing 'system_ext_a'                            OKAY [  0.006s]
Resizing 'system_b'                                OKAY [  0.006s]
Resizing 'vendor_a'                                OKAY [  0.005s]
Resizing 'vendor_b'                                OKAY [  0.006s]
Resizing 'product_a'                               OKAY [  0.005s]
Sending 'product_a' (234012 KB)                    OKAY [433.986s]
Writing 'product_a'                                OKAY [  0.936s]
Resizing 'system_a'                                OKAY [  0.005s]
Sending sparse 'system_a' 1/4 (262140 KB)          OKAY [488.538s]
Writing 'system_a'                                 OKAY [  1.039s]
Sending sparse 'system_a' 2/4 (262140 KB)

dorssel, 2024-04-15:
Did you compile your own kernel? Did you enable "Debug messages for USB/IP" (you shouldn't...)?

See https://github.com/dorssel/usbipd-win/wiki/Troubleshooting on how to obtain more debugging information.

LLeavesG, 2024-04-15:
No, i use the original kernel `Linux lleaves 5.15.146.1-microsoft-standard-WSL2 #1 SMP Thu Jan 11 04:09:03 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux`
and I see the  `/proc/config.gz`

```
CONFIG_USBIP_CORE=y
CONFIG_USBIP_VHCI_HCD=y
CONFIG_USBIP_VHCI_HC_PORTS=8
CONFIG_USBIP_VHCI_NR_HCS=1
# CONFIG_USBIP_HOST is not set
# CONFIG_USBIP_DEBUG is not set
```

I try the trace, but it seems everything is ok.
```
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 172.30.173.224
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.30.173.224
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 2000 ms
info: Usbipd.ConnectedClient[1]
      Client 172.30.173.224 claimed device at 1-10 (USB\VID_18D1&PID_4EE0\08221FDD4005BA).
dbug: 1-10.0[1000]
      Endpoint created
dbug: 1-10.0[1000]
      Endpoint created
dbug: 1-10.0[1000]
      Trapped SET_CONFIGURATION: 1
dbug: 1-10.1[1000]
      Endpoint created
dbug: 1-10.1[1000]
      Endpoint created
trce: 1-10.1[1001]
      Unlinking: PendingSubmits=1, PendingUnlinks=1
trce: 1-10.1[1001]
      Aborting endpoint
```

I noticed when i attach the device to wsl, output some strange code.I don't know if that has anything to do with the problem
```
usbipd attach --wsl --busid 1-10
usbipd: info: Using WSL distribution 'Ubuntu-20.04' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.30.160.1 to reach the host.
WSL wsl: �hKm0R localhost �NtM�n
                                �FO*g\��P0R WSL0NAT !j_
                                                       N�v WSL
WSL N/ec localhost �Nt0
WSL
WSL
```

Now,i use fastboot.exe in wsl2, it works well.

dorssel, 2024-04-15:
The logging from the WSL `usbip` is a bit weird. I've not seen it before (the strange characters are simply terminal codes, probably to set the color). Normally, `usbip` is quiet (i.e., no output).

Is there anything in syslog?

> Now,i use fastboot.exe in wsl2, it works well.

Are you saying the problem does not occur anymore?

LLeavesG, 2024-04-15:
> Now,i use fastboot.exe in wsl2, it works well.

means I used a soft symlink to link the windows fastboot.exe into wsl2 to use it as a fastboot, which is actually still using the windows fastboot.exe.

I checked the syslog

```
Apr 15 11:48:54 lleaves kernel: [49190.447646] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
Apr 15 11:48:54 lleaves kernel: [49190.447683] vhci_hcd vhci_hcd.0: devid(65546) speed(3) speed_str(high-speed)
Apr 15 11:48:54 lleaves kernel: [49190.447701] vhci_hcd vhci_hcd.0: Device attached
Apr 15 11:48:54 lleaves kernel: [49190.794250] usb 1-1: new high-speed USB device number 3 using vhci_hcd
Apr 15 11:48:55 lleaves kernel: [49191.154261] usb 1-1: SetAddress Request (3) to port 0
Apr 15 11:48:55 lleaves kernel: [49192.006621] usb 1-1: New USB device found, idVendor=18d1, idProduct=4ee0, bcdDevice= 1.00
Apr 15 11:48:55 lleaves kernel: [49192.006624] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 15 11:48:55 lleaves kernel: [49192.006624] usb 1-1: Product: Pixel 5
Apr 15 11:48:55 lleaves kernel: [49192.006625] usb 1-1: Manufacturer: Google
Apr 15 11:48:55 lleaves kernel: [49192.006625] usb 1-1: SerialNumber: 08221FDD4005BA
Apr 15 11:51:36 lleaves kernel: [49352.695340] vhci_hcd: unlink->seqnum 3582
Apr 15 11:51:36 lleaves kernel: [49352.695343] vhci_hcd: urb->status -104
Apr 15 11:51:42 lleaves kernel: [49359.006370] vhci_hcd: unlink->seqnum 3585
Apr 15 11:51:42 lleaves kernel: [49359.006373] vhci_hcd: urb->status -104
Apr 15 11:52:02 lleaves kernel: [49378.849341] vhci_hcd: connection closed
Apr 15 11:52:02 lleaves kernel: [49378.849457] vhci_hcd: stop threads
Apr 15 11:52:02 lleaves kernel: [49378.849459] vhci_hcd: release socket
Apr 15 11:52:02 lleaves kernel: [49378.849461] vhci_hcd: disconnect device
Apr 15 11:52:02 lleaves kernel: [49378.849534] usb 1-1: USB disconnect, device number 3
Apr 15 11:52:30 lleaves kernel: [49406.016467] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
Apr 15 11:52:30 lleaves kernel: [49406.016470] vhci_hcd vhci_hcd.0: devid(65546) speed(3) speed_str(high-speed)
Apr 15 11:52:30 lleaves kernel: [49406.016486] vhci_hcd vhci_hcd.0: Device attached
Apr 15 11:52:30 lleaves kernel: [49406.364347] usb 1-1: new high-speed USB device number 4 using vhci_hcd
Apr 15 11:52:30 lleaves kernel: [49406.714311] usb 1-1: SetAddress Request (4) to port 0
Apr 15 11:52:31 lleaves kernel: [49407.556052] usb 1-1: New USB device found, idVendor=18d1, idProduct=4ee0, bcdDevice= 1.00
Apr 15 11:52:31 lleaves kernel: [49407.556059] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 15 11:52:31 lleaves kernel: [49407.556062] usb 1-1: Product: Pixel 5
Apr 15 11:52:31 lleaves kernel: [49407.556063] usb 1-1: Manufacturer: Google
Apr 15 11:52:31 lleaves kernel: [49407.556065] usb 1-1: SerialNumber: 08221FDD4005BA
Apr 15 11:59:10 lleaves kernel: [49806.059881] vhci_hcd: connection closed
Apr 15 11:59:10 lleaves kernel: [49806.059997] vhci_hcd: stop threads
Apr 15 11:59:10 lleaves kernel: [49806.059999] vhci_hcd: release socket
Apr 15 11:59:10 lleaves kernel: [49806.060002] vhci_hcd: disconnect device
Apr 15 11:59:10 lleaves kernel: [49806.060032] usb 1-1: USB disconnect, device number 4
Apr 15 12:04:45 lleaves kernel: [50141.416076] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
Apr 15 12:04:45 lleaves kernel: [50141.416078] vhci_hcd vhci_hcd.0: devid(65546) speed(3) speed_str(high-speed)
Apr 15 12:04:45 lleaves kernel: [50141.416094] vhci_hcd vhci_hcd.0: Device attached
Apr 15 12:04:45 lleaves kernel: [50141.762820] usb 1-1: new high-speed USB device number 5 using vhci_hcd
Apr 15 12:04:45 lleaves kernel: [50141.912770] usb 1-1: SetAddress Request (5) to port 0
Apr 15 12:04:45 lleaves kernel: [50141.963310] usb 1-1: New USB device found, idVendor=18d1, idProduct=4ee0, bcdDevice= 4.19
Apr 15 12:04:45 lleaves kernel: [50141.963312] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 15 12:04:45 lleaves kernel: [50141.963313] usb 1-1: Product: AOSP on redfin
Apr 15 12:04:45 lleaves kernel: [50141.963313] usb 1-1: Manufacturer: Google
Apr 15 12:04:45 lleaves kernel: [50141.963314] usb 1-1: SerialNumber: 08221FDD4005BA
```

LLeavesG, 2024-04-15:
Thank you again!  Maybe i will check later.