dorssel/usbipd-win #924: Dropped bulk transfer ZLP writes through WSL2

kairotronix, 2024-04-19:
I recently spent time reverse engineering the USB protocol for some test equipment so I could control it via Linux.

The protocol in question has the host send a command word through a bulk endpoint, then the host sends a zero length packet through the same endpoint so the device knows to begin processing what is in the buffer.

After emulating the protocol above in PyUSB the device would work in Windows while overwriting the libusb driver, and it would work in native Linux, but it would not work in Windows using USB passthrough to WSL2.

I used Wireshark to analyze the USB packets and noticed that the zero length packets were being dropped when the device was being forwarded to WSL2.

I am not certain it is a usbipd issue, but it seems plausible.

dorssel, 2024-04-19:
Can you make a capture following https://github.com/dorssel/usbipd-win/wiki/Troubleshooting#usb-capture?
And perhaps (considering you seem to know what you are doing) simultaneously a WSL-side USB capture (see https://github.com/dorssel/usbipd-win/issues/300#issuecomment-1066104831)?

kairotronix, 2024-04-20:
I can do the USB capture when I get access to the hardware again on Monday. Unsure if I will have time to do the WSL-side, unfortunately.

karloscarrijo, 2026-10-31:
Sorry for resurrecting an old issue, but I recently got an issue and was able reproduce this with a Poco F3 (Android 13) and adb 37.0.1 in WSL2, latest usbipd-win (5.3.0). A bulk OUT write of 2048 bytes (wMaxPacketSize 512) needs a ZLP, because adbd reads into 16 KiB buffers.

`tcpdump` on WSL, shows `CMD_SUBMIT len=2048` then `CMD_SUBMIT len=0`. Each `RET_SUBMIT` has status 0. The phone then does not respond until any other USB write arrives.

Sending blocks of these lengths, 30 times each:

- 2048, 5120, 8192: 30 of 30 stalled
- 15872, 16896: 15 of 30 stalled
- 16384, 32768, 49152: 0 stalled
- 2049, 5121: 0 stalled

The same adb version, phone, cable and port on Windows with no usbipd passed all of these lengths. 
`adb` on the Linux side stalls with both its native and its libusb backend.

So the ZLP leaves WSL2 and is reported complete, but it does not seem to reach the device. I have no Windows-side capture yet. USB traffic in WSL2 is visible on `loopback0`, and I can share the test script.