dorssel/usbipd-win #927: Update to wiki section

Andryshik345, 2024-04-25:
Please add `bc` and `pahole` packages to initial installing in "Install prerequisites." section of https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-wsl-2-kernel-with-additional-drivers.
I ran into a couple of problems during this process:
[image]
[image]

Also it probably makes sense to add additional `Device Drivers -> USB Support -> USB Mass Storage` feature because this is the main issue that you unable to mount USB stick drive normally and have to recompile the kernel.

dorssel, 2024-04-25:
I'll leave this issue open for others to find. For now, it will remain with the note that is already there:

>Depending on the drivers you will select below, additional packages may be required.

Perhaps I'll find some time in the future for a proper writeup.

bbulkow, 2024-05-08:
I have also spent several hours attempting to mount an EXT4 USB drive, and realized this issue. A NOTE that USB storage is not supported by current WSL as of May 2024, and the steps to compile a WSL kernel to do so, would be greatly helpful. Even if just a note to say it.

Andryshik345, 2024-05-09:
@bbulkow it's already there https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-wsl-2-kernel-with-additional-drivers. You just need to install two additional packages I wrote in the first comment and also add USB Mass Storage feature during menuconfig.