dorssel/usbipd-win #933: Cant Detect devices plugged in as usbipd has them bound

Mark62222, 2024-05-01:
How would i detect a device is plugged in when usbipd has them bounded, i have a method using deviceWatcher bet that doesn't detect the device if its bound to usbipd.

dorssel, 2024-05-02:
Binding in usbipd-win is just an internal administrative (registry) thing. It does not, in any way, alter or even use the device. In other words, it doesn't interfere with anything.

I do no understand what you are asking. You are combining different terminology ("plugged in", "bounded", "detect") together with a reference to some "deviceWatcher" that I am unfamiliar with.

d0n13, 2024-06-04:
> ipd-win is just an internal administrative (

Try `usbipd unbind -a` to get rid of the automatic attach if you had them previously attached to another machine...