dorssel/usbipd-win #935: How to connect WSL2 Ubuntu 22.04 to Windows Joystick

rbhodges, 2024-05-02:
I tried following directions posted for Ubuntu 20.04 but they don't seem to work. Is it possible to send detailed instructions for Ubuntu 22.04 running under WSL2 on Windows 10?

Thanks,
     Richard 
rbhodges@gmail.com

dorssel, 2024-05-02:
`usbipd-win` is distribution independent. What specific instructions are you referring to?

rbhodges, 2024-06-04:
> @rbhodges <https://github.com/rbhodges> / @dorssel
> <https://github.com/dorssel> I'm trying to do the same and have found
> that the WSL2 kernel doesn't support joysticks by default. You'll need to
> rebuild your kernel with different settings. There is a long conversation
> about this in this WSL issue
> I was able to get this working for another user, but I'm still having
> issues since I can see the joystick is mounted, but don't receive events
> from it. But I've been looking for answers and ran into your issue.... so I
> hope you find it helpful.

Thanks. If you get the joystick working, can you please give me details how
you did it?

Thanks,

timmhayes, 2024-06-04:
You could try following what we did in the linked issue, which was cloning the repo and rebuilding your WSL kernel. I'm having issues, but it worked for the other guy.  You could try that right now if you want.

timmhayes, 2024-06-06:
@rbhodges My Xbox controller still isn't working, but my old generic GameSir G3 seems to be functioning now and I'm going to roll with that. If you want to run through the steps I did, read the comment that I posted here:
https://github.com/microsoft/WSL/issues/7747#issuecomment-2143500818

The gist is that I ran through the top steps of of the readme in this [WSL2-Linux-Kernel fork](https://github.com/atticusrussell/WSL2-Linux-Kernel). However that fork is old, so [I made my own and applied the same changes](https://github.com/timmhayes/WSL2-Linux-Kernel). I'm not a Linux expert though, so use it at your own risk. Or better yet, have a look at the changes I made in my last commit. If you follow the directions properly, you can always revert the change by removing the configuration from your Window's `.wslconfig` file. 

I hope this helps.

closed 2025-02-02 by dorssel