dorssel/usbipd-win #936: Auto attach is not immediately stopped after exiting the attach loop

lynxnb, 2024-05-02:
I encountered this issue while trying to implement auto attachment in my [GUI](https://github.com/lynxnb/wsl-usb-manager) for attaching USB devices to WSL (yes, another one).

Running `usbipd attach --wsl --auto-attach --busid X-X` and then exiting the loop either by Ctrl-C or closing the terminal window does not immediately stop the auto attach loop.  
Trying to `usbipd detach --busid X-X` after exiting the loop results in the device being reattached one last time, although the loop was already exited. Trying to detach one more time finally detaches the device.

I'm not sure why this is happening, or if there is any fix for this since attaching is a client side thing and the `attach --wsl` command is only a convenience method, but I'd like to get some feedback.

dorssel, 2024-05-02:
Yes, that makes sense. The (windows side) wsl.exe that is used to run a command on the linux side has no real way of *stopping* the process. In fact, when you kill the windows wsl.exe command, the linux command that it started actually keeps running.

This is where we try to stop everything: https://github.com/dorssel/usbipd-win/blob/master/Usbipd/Wsl.cs#L135-L164

The comments should provide enough rationale.

lynxnb, 2024-05-03:
Right, that makes sense, thanks for the quick response! I'll go ahead and close the issue since there's nothing to be done here.

closed 2024-05-03 by lynxnb