dorssel/usbipd-win #943: Failed to connect ThorLabs

waswis, 2024-05-06:
Hi, 

I tried to attach ThorLabs PM100USB from windows to WSL2 but seems like I'm going to a dead end. For some reason, I cannot see PM100USB connected to ubuntu using lsusb even in powershell it says that the device is attached to WSL. Can you help me with this?

Many thanks,
Wastu

dorssel, 2024-05-06:
What is the output of (WSL-side) `lsusb` after attaching?

waswis, 2024-05-07:
[image]

waswis, 2024-05-08:
No, I'm using the same account for admin and local account (checked under control panel -> user account)

dorssel, 2024-05-08:
> in powershell it says that the device is attached to WSL

Which command? What is the output? Which version of `usbipd-win` are you using?

waswis, 2024-05-10:
Hi!

This is the version of the usbipd-win I am using: 4.2.0+54.Branch.master.Sha.5b3...

I want to attach the PM100USB to WSL (already shared using bind)
[image]

After attaching using attach
[image]

Checking on my WSL
[image]

dorssel, 2024-05-10:
Looks like the device is detaching immediately again. Can you follow the troubleshooting at https://github.com/dorssel/usbipd-win/wiki/Troubleshooting?