dorssel/usbipd-win #944: [Question] Use automation dll to attach all devices that are connected+bound but not attached

daantimmer, 2024-05-06:
I am trying to use the automation dll (`Get-UsbIpdDevice`) to be able to attach all devices that are currently connected _and_ bound, but not attached. But I am unable to go from 'Get-UsbipdDevice' context to something that can invoke `usbipd attach --wsl --hardware-id ???`.

```
Get-UsbipdDevice | Where-Object {$_.IsConnected -and $_.IsBound -and -not $_.IsAttached} | <what do I put here to invoke> | usbipd attach --wsl --hardware-id <hardwareid from cmdlet entry/ies>
```

dorssel, 2024-05-07:
```
Get-UsbipdDevice | Where-Object {$_.IsConnected -and $_.IsBound -and -not $_.IsAttached} | ForEach-Object { usbipd attach --wsl --busid $_.BusId }
```

daantimmer, 2024-05-29:
Thank you! This is the script I ended up with:

```pwsh
usbipd list

Import-Module $env:ProgramW6432'\usbipd-win\PowerShell\Usbipd.Powershell.dll'

Get-UsbipdDevice | Where-Object {$_.IsConnected -and $_.IsBound -and -not $_.IsAttached} | ForEach-Object { usbipd attach --wsl --busid $_.BusId }
```