dorssel/usbipd-win #947: "Incompatible Hub" when using SEH UTN Manager (network USB hub)

alog243, 2024-05-10:
Issue: When using SEH UTN manager to activate USB devices on Windows 10, the device state is listed by usbipd as "Incompatible Hub" and cannot be used:

```
> usbipd list
Connected:
BUSID VID:PID    DEVICE                     STATE
      XXXX:XXXX  USB Serial Converter       Incompatible Hub
2-3   XXXX:XXXX USB Mass Storage Device     Not shared
...
```
Maybe this is expected behavior, since these devices are activated via UTN Manager's driver and not on the physical USB bus? If there is a workaround please let me know.

dorssel, 2024-05-10:
Probably the (OEM) USB hub driver reporting incompatible location information for the device. Try to find the device in Device Manager and post a screenshot of this page:

[image]

Also, set the Device Manager to "By connection", and locate the hub that the device is connected to, like so:

[image]

Then, display the drivers *of the hub*, like so:

[image]

alog243, 2024-05-10:
Unfortunately I can't post screenshots as it's on a private network, but I can type contents:

Device Manager:
```
> UTN Host Controller
    > UTN Root Hub
        > USB Serial Converter
            > USB Serial Port (COM8)
```

USB Serial Converter properties (location definitely looks strange, as you suspected):
```
[General]
USB Serial Converter

Device type: Universal Serial Bus controller
Manufacturer: FTDI
Location: SEH myUTN;TTL232R-3V3
```

UTN Root Hub properties:
```
[Driver]
UTN Root Hub

Driver Provider: SEH Computertechnik GmbH
Driver Date: 8/30/2023
Driver Version: 4.0.0.0
Digital Signer: Microsoft Windows Hardware Compatibility Publisher

...
[Driver File Details]
UTN Root Hub

Driver files:
C:\WINDOWS\system32\DRIVERS\sehutn.sys

Provider: [etc, same as details above]
```

dorssel, 2024-05-10:
I don't know if they are compatible, but you could try to install the stock Microsoft hub drivers for "UTN Root Hub". Hopefully the "UTN Host Controller" (for which you definitely do need the OEM drivers) exposes the hub as a standard, generic hub. If so, replacing the drivers of the hub with the Microsoft ones will make the device work. If not, then there is nothing that can be done (as in that case it is, in fact, an incompatible hub).

alog243, 2024-05-10:
I'll give that a shot when I'm back in the office next week

alog243, 2024-05-13:
Tried using a couple different generic USB Hub drivers, none of them seemed to work, Windows failed to load any drivers for the device other than `sehutn.sys`

dorssel, 2024-05-13:
Then I'm afraid that device is not going to work with `usbipd-win` unless you can connect it to another (compatible) hub.

closed 2024-05-14 by dorssel