dorssel/usbipd-win #949: Rebooted from a bug check

GreatAg, 2024-05-11:
Hi when i attach to a bound usb, in server i recieve Rebooted from a bug check error.
How can i prepare a detailed log for you to debug what is goes wrong?
Thanks.

dorssel, 2024-05-11:
A bug check as in Blue Screen of Death (BSOD)?
It should be in the Windows System Event Log. And on the BSOD itself it should list the kernel module (probably a driver, XXX.SYS).

GreatAg, 2024-05-12:
Yes, it caused BSOD.
My windows is 10, do you know what is wrong? does usb/ip sensitive to any programs?

dorssel, 2024-05-12:
It's a driver, but I cannot tell you which one. It is somewhere in your Windows logs. `usbipd-win` is a user mode program and therefore is not the cause of the BSOD.

Andre-Gloria, 2024-06-03:
Hi, I was having the same issue, it was so bad it was unusable, using the _--force_ option of bind solved it for me.
Example: usbipd.exe bind --hardware-id xxxx:XXXX --force

closed 2025-02-02 by dorssel