dorssel/usbipd-win #955: can you add manually vendorid and productid of unknown devices?

comsyspro, 2024-05-18:
i have some usb devices which are not listed in the database for vid and pid.  pairs. so where can i find these vid and pid list and are these values stored in a file and can you add your own entries for unknown vids and pids? from where usbipd is getting these values?

dorssel, 2024-05-20:
`usbipd-win` gets the device descriptions from Windows. The `list` command has an option to display the Linux description instead (`usbipd list -u`). Those are taken from a copy of the database: http://www.linux-usb.org/usb-ids.html
You can add your device there, if you want, and in some future update `usbipd-win` will automagically pick it up.

comsyspro, 2024-05-21:
so i could also update this file with my vid and pid?
https://github.com/dorssel/usbipd-win/blob/281b97466e8270f2117ccfcd3e7a827829c9011d/Usbipd.Automation/usb.ids
do i have to compile then or can usbipd read this list from an external file so you don't need to compile the code?

comsyspro, 2024-05-21:
thanks for your info. it would be nice if one can add an adapted usb.id file in basefolder of usbipd and let usbipd detect if this file exists and if yes load this file and if not use the compiled usb.id instead. would that be possible to do?

dorssel, 2024-05-21:
Possible, yes. But I've already decided against that earlier. See https://github.com/dorssel/usbipd-win/pull/468. All about performance which sucks on .NET when reading a file like this ... multiple seconds on systems with virus scanners ... So, sorry, that is not going to happen.