dorssel/usbipd-win #959: Trying to restore iPhone from WSL - asyncReadComplete, message was too large

d0n13, 2024-05-23:
Hi @dorssel 

I am trying to run an [idevicerestore ](https://github.com/libimobiledevice/idevicerestore) in wsl to a connected iPhone. When the firmware starts to load onto the phone we get the following error in syslog as reported by usbmuxd on the WSL side.

```
2024-05-23T16:41:42.806058+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.805][3] usbmuxd v1.1.1 starting up
2024-05-23T16:41:42.806434+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.806][3] Successfully dropped privileges to 'usbmux'
2024-05-23T16:41:42.806483+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.806][3] Using libusb 1.0.27
2024-05-23T16:41:42.808529+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: libusb: warning [op_get_configuration] device unconfigured
2024-05-23T16:41:42.808604+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: libusb: error [op_get_active_config_descriptor] device unconfigured
2024-05-23T16:41:42.808635+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.808][3] Could not get old configuration descriptor for device 1-21: LIBUSB_ERROR_NOT_FOUND
2024-05-23T16:41:42.850999+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.850][3] Initialization complete
2024-05-23T16:41:42.851142+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.850][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
2024-05-23T16:41:42.852935+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.852][3] Connecting to new device on location 0x10015 as ID 1
2024-05-23T16:41:42.853395+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.853][3] Connected to v2.0 device 1 on location 0x10015 with serial number 00008101-001E48CXXXXXXXXX
2024-05-23T16:41:48.692940+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[36mINFO#033[0m Daemon: connecting to Windows Agent
2024-05-23T16:41:48.693071+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[37mDEBUG#033[0m Updated systemd status to "Connecting"
2024-05-23T16:41:48.693316+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[33mWARNING#033[0m Daemon: could not connect to Windows Agent: could not get address: could not read agent port file "/mnt/c/Users/donie/.ubuntupro/.address": open /mnt/c/Users/donie/.ubuntupro/.address: no such file or directory
2024-05-23T16:41:48.693382+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[36mINFO#033[0m Reconnecting to Windows host in 60 seconds
2024-05-23T16:41:48.693406+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[37mDEBUG#033[0m Updated systemd status to "Not connected: waiting to retry"
2024-05-23T16:41:49.409533+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:49.409][1] device_control_input: ERROR (on device): asyncReadComplete, message was too large (65536 bytes, max = 65535)
2024-05-23T16:41:49.411258+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:49.411][1] device_control_input: ERROR (on device): asyncReadComplete, message was too large (65536 bytes, max = 65535)
2024-05-23T16:42:23.510946+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:42:23.510][2] device_control_input: Got unhandled payload type 4
2024-05-23T16:42:48.752655+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[36mINFO#033[0m Daemon: connecting to Windows Agent
2024-05-23T16:42:48.757790+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[37mDEBUG#033[0m Updated systemd status to "Connecting"
2024-05-23T16:42:48.757865+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[33mWARNING#033[0m Daemon: could not connect to Windows Agent: could not get address: could not read agent port file "/mnt/c/Users/donie/.ubuntupro/.address": open /mnt/c/Users/donie/.ubuntupro/.address: no such file or directory
2024-05-23T16:42:48.757878+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[36mINFO#033[0m Reconnecting to Windows host in 60 seconds
2024-05-23T16:42:48.757888+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[37mDEBUG#033[0m Updated systemd status to "Not connected: waiting to retry"
```

The [idevicerestore](https://github.com/libimobiledevice/idevicerestore) command logs are shown below for reference ...

Is it possible to fix this "**message was too large (65536 bytes, max = 65535)**"??

Many thanks
Donie

**Note**: during the process of restoring the firmware the iPhone presents as different things on the usbipd side as shown. The apple devices persisted are all the same device. Just in case it's relevant. I have to manually bind and attach during the process before the code times out. I'm not worried about that as we can automate that later. Just need it to transfer the image to restore manually first.

```
PS C:\Users\donie> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-1    05ac:12a8  Apple Mobile Device USB Composite Device                      Attached
2-6    1bcf:28cf  Integrated Webcam                                             Not shared
2-8    0a5c:5842  Dell ControlVault w/o Fingerprint Sensor, Microsoft Usbcc...  Not shared
2-10   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared
3-2    0bda:8153  Realtek USB GbE Family Controller                             Not shared
4-3    046d:c52b  Logitech USB Input Device, USB Input Device                   Not shared
4-5    0bda:4014  Realtek USB Audio                                             Not shared

Persisted:
GUID                                  DEVICE
33a1cdbf-4373-4d17-a813-00d3bb825d3f  Apple Recovery (iBoot) USB Composite Device, Apple Mobile...v
5002e98e-b231-464b-8fb7-72c912858934  Apple Recovery (iBoot) USB Composite Device, Apple Mobile...v
```

```
$ idevicerestore --erase iPhone13,2,iPhone13,3_17.5_21F79_Restore.ipsw
idevicerestore 1.0.0-188-g56d2c01
Found device in Recovery mode
ECID: 8524247579XXXXXX
Identified device as d53gap, iPhone13,2
Device Product Version: N/A
Device Product Build: N/A
Extracting BuildManifest from IPSW
IPSW Product Version: 17.5
IPSW Product Build: 21F79 Major: 21
INFO: device serial number is FFXDNXXXXXXX
Device supports Image4: true
[... 140 lines cut ...]
Personalizing IMG4 component DeviceTree...
Extracting liquiddetect@2532~iphone-lightning.im4p (Firmware/all_flash/liquiddetect@2532~iphone-lightning.im4p)...
Personalizing IMG4 component Liquid...
Extracting lowpowermode@2532~iphone-lightning.im4p (Firmware/all_flash/lowpowermode@2532~iphone-lightning.im4p)...
Personalizing IMG4 component LowPowerWallet0...
Extracting lowpowerfindmymode@2532~iphone-lightning.im4p (Firmware/all_flash/lowpowerfindmymode@2532~iphone-lightning.im4p)...
Personalizing IMG4 component LowPowerWallet1...
Extracting recoverymode@2532~iphone-lightning.im4p (Firmware/all_flash/recoverymode@2532~iphone-lightning.im4p)...
Personalizing IMG4 component RecoveryMode...
Extracting WirelessPower.iphone13.im4p (Firmware/WirelessPower/WirelessPower.iphone13.im4p)...
Personalizing IMG4 component WCHFirmwareUpdater...
Extracting iBoot.d53g.RELEASE.im4p (Firmware/all_flash/iBoot.d53g.RELEASE.im4p)...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.d53g.RELEASE.im4p (Firmware/all_flash/sep-firmware.d53g.RELEASE.im4p)...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.d53g.RELEASE.im4p (Firmware/all_flash/sep-firmware.d53g.RELEASE.im4p)...
Personalizing IMG4 component SEP...
Sending NORData now...
ERROR: Unable to send NORData
ERROR: Unable to send NOR data
ERROR: Unable to successfully restore device
ERROR: Unable to restore device
$

```

d0n13, 2024-05-23:
Maybe related to [#867](https://github.com/dorssel/usbipd-win/issues/867)

dorssel, 2024-05-24:
The usbmuxd error is logged here: https://github.com/libimobiledevice/usbmuxd/blob/1.1.1/src/device.c#L601

It looks like it is not usbmuxd's error. usbmuxd is just reporting/relaying the error from whatever underlying driver/library it is talking to. But I don't know what that is; I have no Apple knowledge.

The explanation in https://github.com/dorssel/usbipd-win/issues/867#issuecomment-1956410565 is still the best I can come up with. Try to find whatever usbmuxd is using to talk to the device and see if you can increase the buffer size there.

d0n13, 2024-06-05:
> The explanation in [#867 (comment)](https://github.com/dorssel/usbipd-win/issues/867#issuecomment-1956410565) is still the best I can come up with. Try to find whatever usbmuxd is using to talk to the device and see if you can increase the buffer size there.

I changed all buffer sizes I could find in idevicerestore which had hardcoded buffers of 65535 and 0xFFFF but the error was the same... Will have to dig deeper for this one I think.